nasa-jpl/open-source-rover #1: Git LFS storage is above quota

dmittman, 2018-06-30:
@vssystemluba, could you take a swing through the repo and compress the PDF files? As they are, they're ±40MB, and if compressed, they'd be ±2MB. The extra storage has consumed our Git LFS quota (which I didn't know about until I was notified). We'll look into how we can increase our available Git LFS storage space for our org, but until then, compressing the large PDFs will help. Happy to do that in a pull request. I simply use macOS Preview, and Save As using the Reduce File Size Quartz Filter.

vssystemluba, 2018-06-30:
@dmittman after some digging I'm pretty convinced that there aren't any .pdf files from the repository being stored in Git LFS, only .STEP and .IGS files.  I've done a sweep through the repository and compressed all of those files, which resulted in small enough file sizes that they could be removed entirely from Git LFS.  Please let me know if the quota issue has been resolved.

vssystemluba, 2018-07-10:
@dmittman can you confirm that the org's Git LFS quota has been freed up by my actions above?  If so, I will close this issue.

Thanks!

vssystemluba, 2018-08-02:
Marking this as closed since I believe we are no longer storing any files in Git LFS.  @dmittman if you see further issues feel free to re-open.

closed 2018-08-02 by vssystemluba

dmittman, 2018-08-03:
I did a little more research and found this:

> After you remove files from Git LFS, the Git LFS objects still exist on the remote storage and will continue to count toward your Git LFS storage quota.

From https://help.github.com/articles/removing-files-from-git-large-file-storage/#git-lfs-objects-in-your-repository

Unfortunate, if true. I think I can get GitHub.com support to delete the remote storage, which should fix the problem without needing to delete and recreate the repo.