nasa-jpl/open-source-rover #2: Licensing concern!

accidentalmeme, 2018-08-02:
The file LICENSE.txt contains the Apache 2.0 license. However, the file DISCLAIMER.txt says the rover cant be modified, distributed, or used commercially without prior approval. This is in conflict with the Apache 2.0 license.

NatTuck, 2018-08-02:
Looks like the license is written approval for commercial use. There's no conflict, the license just overrides the disclaimer.

jdalrymple, 2018-08-03:
Wouldn't it make more sense to remove that section of the desclaimer to avoid future confusion ?

accidentalmeme, 2018-08-03:
@NatTuck While that may be correct, we don't want uncertainty. In addition, by downloading you agree to both the Apache 2 license and the disclaimer.

vssystemluba, 2018-08-06:
I'm working with the legal team to get this resolved in a way that's more clear and concise.  Thanks for bringing this to our attention.

chrismattmann, 2018-08-09:
hi everyone, I manage our Advanced IT Research and Open Source office at NASA JPL. Our position is that the license chosen overrides the disclaimer. We have approval to license this software under the Apache License, version 2, which allows others to do what they wish (even build commercial software on it). That should supersede the attached disclaimer which are artifacts of an evolving OSS release process that we are trying to improve.

You may cite this comment should the need arise, which I doubt it will (ever). Enjoy the OSS rover and great work @vssystemluba and team!

closed 2018-08-09 by vssystemluba