nasa-jpl/open-source-rover #13: Stage 7 upgrades

ridhwaans, 2018-08-05:
In the readme, what upgrades are nasa referring to in stage 7? RPi shields? maybe RPi has sensors to emulate some of the Curiosity instruments https://en.wikipedia.org/wiki/Curiosity_(rover)#Instruments

Naphier, 2018-08-05:
I think this is something the project really needs contribution to. Building a rover is entry-level robotics. Integrating sensors for interesting autonomous behavior can be a bit more involved. Really needs some sonar for collision detection, IMU for roll over warnings, camera with gimbal + OpenCV for object tracking. Environmental sensors might be nice too (temp, pressure at least). The use of 13 GPIO pins might be a bit of a limiting factor here, but it may be possible to off-board the sensors to an Arduino connected to the USB ports. Another possibility is to get an SSC-32U and Sabretooth 2x12 (might need 2 since this is a 6-wheeler) board for driving the rover via USB to the Pi, that would also free up GPIOs. That would also allow addition of something like a Lynxmotion AL5D robotic arm(s) to the bot.

The chassis of this bot is neat, but if you want to just make a rover, there are much cheaper options.

vssystemluba, 2018-08-06:
Good discussion.  I've added some example upgrades in the master README.md.

closed 2018-08-06 by vssystemluba