nasa-jpl/open-source-rover #14: repeated vertical scale

Dmole, 2018-08-05:
"Maximum 90 deg vertical scale = 12 [in]"

Stair climbing would be "repeated 7 3/4 inches vertical / 10 inches horizontal", is that achievable without modification?

vssystemluba, 2018-08-17:
I had a chance to play with the rover this evening and set up the situation you're talking about.

The short story is that the rover struggled with this setup and without modification, I will say that **no, the rover doesn't quite climb this setup**.  ...but it's VERY close...

Now for the longer story:

The rover gets up and over the first 7 3/4 inch vertical wall no problem, as expected:

[All wheels on ground](https://drive.google.com/open?id=1aebdlawkZKM-jNjU0fzTDqC7_4nwjCc6) -> [First wheels successfully on top of first step](https://drive.google.com/open?id=1GjM1yernux2WxN-n5A5KJKNDvtNim-TP)

After that, we run into issues.  The main problem with this geometry is the 10-inch horizontal spacing of the 'stairs'.  This distance is similar enough to the distance between the rover's front two wheels and its middle two wheels, so when the rover hits the second stair, 4 wheels are all trying to scale vertically at the same time:

[Starting to get stuck on second stair](https://drive.google.com/open?id=17c1U1O1yrmhioDXuMzbnlzaJCcJrzgwI)

You can see a video of this process here: https://drive.google.com/open?id=1dMNV0npAofhaYSRp0pE2AZPY-sfsrf26

Now for the even longer story...  JPL's rover is actually equipped with a 'boost' button which doubles the speed that we instruct the wheels to drive at (because why not!).  When I hit that button when the rover is in the "stuck on 2 stairs" state, the rover looked like it WAS able to get up the next stair, but once the back wheels pull into the stair, the angle of the stairs is so steep that the rover actually tipped over backwards.  I don't have good video of this so you'll have to take my word on it.

In conclusion, the rover can't climb these stairs unmodified.  The rocker-bogie isn't optimized for climbing stairs; it's optimized for getting over a rocky desert-like environment like that we see on Mars.  I have no doubt that changing the leg lengths and/or the rocker-bogie geometry and swapping out your wheels for some wheels with other materials that grip better on your specific surface would yield you better results for your specific stair-climbing use case.

closed 2018-08-17 by vssystemluba

Dmole, 2018-08-17:
Thanks for looking into this.

vssystemluba, 2018-09-13:
For posterity, I've added the above video and photos to our flickr page (https://www.flickr.com/photos/160731951@N05/) in case the above links become outdated.