nasa-jpl/open-source-rover #19: Request for any documentation on software structure

n1xnx, 2018-08-06:
I know how exactly much fun it is to write documentation, but it would be very helpful if the developers can add some text or diagrams describing the organization and structure of the rover software. In particular, I'd be interested in how the motor control layer works, as I might like to implement a different hardware controller for brushless DC motors. Thank you, developers, for this awesome and impressive project.

ericjunkins, 2018-08-06:
I'll throw something up this week when I get a little bit of time to create that, good suggestion

pabloriveracelerity, 2018-09-14:
Note that Im currently writing unit tests for the rover software over and that will shed some good light into the how it works @n1xnx 

Code is here: https://github.com/nasa-jpl/osr-rover-code

vssystemluba, 2019-07-26:
With the new ROS rework, we've added some documentation about the new software.  It's available at https://github.com/nasa-jpl/osr-rover-code/tree/68e451cb5454ea2087485b4e5718524bba398aa0!

closed 2019-07-26 by vssystemluba