nasa-jpl/open-source-rover #21: Radiation Hardening

gosha1128, 2018-08-06:
This is a great project.  

I realize this rover is not meant for actual space travel, but I'm curious what it would take to actually get one on Mars.  

Isn't there a lot of added expense to harden it for radiation ?  How much would that cost ?

vssystemluba, 2018-08-06:
For reference (and for an idea of how difficult it is to radiation harden something), the main compute element on the Mars 2020 rover (still being built to launch in 2020 and is the current state of the art) is a PowerPC RAD750 (https://en.wikipedia.org/wiki/RAD750).  The max CPU clock rate is about 200MHz and the rover will have 256MB of RAM with 2GB of flash memory.  It's insanely difficult to radiation harden (much less flight test/approve) spacecraft compute hardware, and there's no AAA on Mars in case something needs fixing.  ;)

closed 2018-08-06 by vssystemluba

gosha1128, 2018-08-07:
Folks - Thanks for all the great information on rad-hardening !

George

Phoenix1747, 2018-08-07:
Space-hardening is really interesting nevertheless extremely hard. Trying to make a "space-approved" version of the Raspberry Pi is most likely not possible at all I guess, but what's much more interesting and exciting (and also off-topic, sorry) is making a space hardened version of a MCU like a standard ATmega328P or other more powerful ones, where you can design the system around the bare microcontroller and try to shield it against radiation and so on. Must be incredibly hard but sounds hella interesting.