nasa-jpl/open-source-rover #22: Testing scenarios

shankariyerr, 2018-08-07:
What other testing scenarios have been used opposed to a pile of rocks? What about gravitational and environmental factors?
- [Ash Pearson on ProductHunt](https://www.producthunt.com/posts/open-source-rover#comment-645086)

vssystemluba, 2018-08-07:
This isn't an "issue" with the repository or the source code, per say, so I'm going to close this issue and suggest asking questions like this on our [community forum](http://s15.zetaboards.com/JPL_OpenSource_Rover/index/) in the future.

However, here's a brief answer that I'll provide here:

We have done a fair amount of driving the OSR around various common terrain on lab.  This includes streets, curbs, stairs, grass, dirt, and surfaces with small rocks (less than about 8in diameter).  We have also driven the rover for brief stints in our "Mars Yard" where we do testing for our actual rovers.  We've successfully climbed up curbs, over ~6in rocks, over people laying on the ground, etc. However, we have NOT done quantitative testing of the climbing capabilities of this rover.  The project was made to be built fast and inexpensively and then released as a base model.  If you desire extensive quantitative testing and hard numbers for the climbing capabilities of the rover, I suggest that you join the community of builders and contribute the results of those tests yourself.  :)

All that said, the short story is that the rover typically does fairly well climbing over "reasonably-sized" obstacles.

closed 2018-08-07 by vssystemluba

shankariyerr, 2018-08-07:
Thanks a lot @vssystemluba for taking your time to explain!