nasa-jpl/open-source-rover #26: Drive Motor Clamp - Mechanical over-constraint

ericjunkins, 2018-08-16:
## Problem 
The attachment scheme for the drive motors leads to a mechanical over-constraint.

[image]

Here we can see that the clamping force on the aluminum 25mm Diameter motor clamp will be over-constrained when bolting the clamp into the aluminum U-channel

[image]

The red arrows show the force put on the piece from the clamping screw, and green arrows show force put on by bolting to the u-channel. These force are in direct opposition, and overtime lead to one of two things:

 * The motor loosens inside the clamp and is allowed to free spin, which will tangle and eventually break your wires
 * The clamp loosens inside the u-channel and the wheel assembly becomes loose to the rest of the robot chassis

## Solution
The motor should mount on using the bolting pattern at the bottom of the motor, similar to the corner motor mounting scheme. This will alleviate this over constraint and allow all pieces to be tightened properly 

[image]

siero89, 2018-09-20:
what about if instead using a clamping screw u use a magnetic clamp ?

MisterUniverse, 2018-10-19:
Using a cotter pin at the end of the bolt will keep it from coming loose

ericjunkins, 2020-06-25:
Will be updating documentation for the screw tightening order to properly take up mechanical tolerances

closed 2023-09-25 by Achllle