nasa-jpl/open-source-rover #27: Clamping Hub: Mechanical over-constraint

ericjunkins, 2018-08-17:
## Problem

There is a mechanical over-constraint imposed by the 0.25 inch clamping hub used to attach the corner wheel assemblies to the corner steering assemblies. 

[image]

This over-constraint is caused by the opposition of the clamping action of the clamping hub around the 1/4 inch shaft, and the bolting force attaching it to the aluminum U-channel

[image]

The force caused by the clamping hub is shown in red, and force from the bolting to the U-channel is shown in green. There is not enough tolerance in the clamping hub to both very tightly grip around shaft and bolt to the u-channel.

## Solution

In order to solve this issue you need to find a way to de-couple the attachment of the hub from the clamping force on the 0.25 in shaft. This could be done in many different ways, a few potential solutions would be:

 * Find a way to split this piece up into multiple pieces that are not coupled rigidly in those directions
 * Find a way to increase the mechanical tolerance in the bolting so that these forces are not in opposition

JHPHELAN, 2018-09-18:
What if you just left off one of the "front" vertical screws on the clamp out?  The clamp would remain held by 3 screws while allowing the horizontal tightening screw to pull the free half closer.  I'm no engineer so tell me if this is wrong-headed!
Jim Phelan, Mars Rover project lead, **Houston Robotics**

ericjunkins, 2020-06-25:
Will be updating documentation for the screw tightening order to properly take up mechanical tolerances

closed 2023-09-25 by Achllle