nasa-jpl/open-source-rover #28: Shaft Coupler -

ericjunkins, 2018-08-17:


davidadkins1, 2018-08-24:
Has anyone considered this? https://www.amazon.com/Stainless-Steel-Screw-Shaft-Coupler/dp/B00KQRRVHM

vssystemluba, 2018-08-24:
@davidadkins1 we used shaft couplers that were extremely similar to those ones in our first design of the rover.  We found that no matter how hard we torqued down those set screws, they would always back themselves out and lose grip on the shafts.

ericjunkins, 2018-08-24:
Initially we were using almost the exact part, however the torque seen on the corner system was higher than what the set screw was able to hold, and inevitably the set screws would loosen and back themselves out. Even with loctite this was the case eventually

vssystemluba, 2018-08-24:
On a semi-related note, I know some roboticists use motors hex drive shafts.  Without doing much digging, I found these... http://www.revrobotics.com/rev-41-1301/.  If we found a way to mount a motor with a hex drive shaft, we might be able to solve the shaft coupler problem.

ericjunkins, 2018-09-18:
I'm working on a potential 3D printed solution to this. I have designed and done a very quick test using an ABS based material, with good initial results.

**Proposed Solution**
[image]

**Motor Mounting Hub:**
[image]

Mounts the motor slightly higher up, but gives a more convenient access to the two screws that bolt into the motor. Instead of having to disassemble the entire coupler-standoff system to get to them the whole motor mount just uses the 4 bolting holes used for the pillow bearing standoffs as well. 

**Shaft Coupler:**
[image]

This coupler takes advantage of the D-profile on both of the shafts, which will be the mating face for torque transmission. In addition with the decrease in outside diameter the upper half of the coupler can be extended to the entire length of the 4mm motor shaft, giving increased contact area. Set screws with heat set inserts will be added to handle axial loads only.

I'm going to test with ABS first and then afterwards test with Poly-Carbonate ABS well for increased hardness on the coupler.

vssystemluba, 2019-06-13:
Based on this discussion, the discussion and eventual solution on the OSR forum at https://www.tapatalk.com/groups/jpl_opensource_rover/shaft-coupler-solutions-discussion-t69.html, and @kevincloutier's pushed solution (9b407a6), I am closing this issue as solved.  Thanks everyone for the input and the design improvement!

closed 2019-06-13 by vssystemluba