nasa-jpl/open-source-rover #30: Potential Parts QTY inconsistency

plingafelt, 2018-08-19:
Really looking for confirmation: I have just started reviewing osr_Master_parts_list.xlsx to prepare for this project.  I expect column C (Qty in project) is the number used in the overall project (the sum of the sub-assembly part lists).  In some instances this number does not match column B (Qty to Buy).  I would expect this if the # required represents a partial consumption of a bulk order (line #34: 72 screws needed that are sold in lots of 100 - Qty to Buy = 100; Qty in project =72).  In some cases these numbers do not agree (in one case the, line 79, the Qty in project > Qty to Buy).  Can some one clarify is this is a data error or is my understanding of the two columns incorrect?  Thanks.
[image]

ericjunkins, 2018-08-20:
Yes that is indeed a data error, and will need to be looked into as far as what is the correct quantity necessary. The reason there are some mismatches in those columns even if it not sold in qty 1 would be to have extras of components that I thought were cheap enough to warrant having a few extra just in case, such as the standoffs in line 79. My guess is 16 are needed in project, 20 is suggested, but I will have to dig into this to verify qtys.

vssystemluba, 2018-09-13:
This inconsistency will also likely be resolved once we merge PR #16

vssystemluba, 2019-03-12:
Resolved by e243445

closed 2019-03-12 by vssystemluba