nasa-jpl/open-source-rover #33: Offer FreeCAD fcstd files as an alternative to Solidworks

luzpaz, 2018-08-23:


ericjunkins, 2018-08-23:
There are step files and igs files provided as alternatives to Solid works parts

luzpaz, 2018-08-23:
If someone took the time to convert said step and igs files to FCstd files, would you consider including them in this repo ?

ericjunkins, 2018-08-23:
I'm unfimiliar with that file type. What benefit does that type have beyond igs and step files? By file size the CAD already takes up a huge amount of space so it's prefer not to just keep adding CAD file types unless there's a compelling argument

vssystemluba, 2018-08-23:
Agreed.  FCStd is basically a .zip file that includes a STEP file plus a lot of other things that we don't really need or use for this project, so I'm hesitant to include those files in the repository.  FreeCAD can import STEP files natively, so if anyone does need an FCStd file, FreeCAD would be an easy way to create one from the contents of this repository.

closed 2018-08-23 by vssystemluba