nasa-jpl/open-source-rover #36: Rocker-Bogie Assembly spacing issue

xtream1101, 2018-09-08:
In Section 2, Step 2 of the Rocker-Bogie Assembly, when using the T1 (1/4") standoffs it creates to much of a gap it seems, also need to use B4 (5/8" screws) to connect it. I would think that the size of the standoff is incorrect and should be a smaller size. Is this correct?

[Photo of the spacing with 1/4" standoffs](https://photos.app.goo.gl/Rk7QB4n5XqdCiLCX9)

ericjunkins, 2018-09-10:
The 1/4" spacer is the correct length of spacer for the distance, but the 5/8" should be the screw used, it says 3/4" right now. 

[image]

[image]

ericjunkins, 2018-09-13:
Just following up on this, did you end up still seeing an issue here? Figuring out if this needs to be looked into further

xtream1101, 2018-09-14:
It seemed to fit together fine. Things started to line up better after the whole thing was assembled.

closed 2018-09-20 by ericjunkins