nasa-jpl/open-source-rover #38: Where is the source code for the JavaScript simulator?

cirosantilli, 2018-09-14:
The simulator at https://opensourcerover.jpl.nasa.gov/#!/home

I won'd build this thing, but I might hack its simulation for ML tests.

You guys are awesome.

vssystemluba, 2018-09-27:
The source code for the website is pure javascript, so in theory you can download the source that your browser is showing you to view the code that generated the website.  I believe we cannot distribute the un-minified code due to the agreement we laid out with the contractor who built the site for us.

Even if we could share that code, I believe it would take SUBSTANTIAL development effort to try and add a simulated camera for somebody who is trying to modify the existing codebase.  If you want to discuss in more detail, I suggest reaching out the contractor who built the site (Active Theory).  They published a case study on the site at https://medium.com/@activetheory/nasa-jpl-open-source-rover-9d2b8790beaf which may give you some additional useful information.

Good luck and thanks for your interest in the Open Source Rover!  :)

closed 2018-09-27 by vssystemluba

cirosantilli, 2018-09-28:
> The source code for the website is pure javascript, so in theory you can download the source that your browser is showing you to view the code that generated the website. I believe we cannot [...]

OK, thanks for confirming. Maybe find a less pick contractor next time :-)

As a wise person once said, there is no closed source for those who know assembly. But I'll just stick to Gazebo + Bullet + OpenGL for now :-)