nasa-jpl/open-source-rover #39: B10 screw incorrect

JHPHELAN, 2018-09-16:
The Master Parts List lists B10 as a M2.5 x 4mm cap head screw.
The Electronics Board parts list shows 11mm.
For consistency (gasp!) it should be a button head screw to match the others on the board.
McMaster only has even sizes in M2.5 button head hex screws, so 10mm: 
https://www.mcmaster.com/92095a460

JHPHELAN, 2018-09-17:
On further exploration, the T7 M2.5 standoff is listed as 11mm long in the build but 12mm long on the Master Parts List and in the McMaster link.  Longer is better.  But the B10 M2.5 screw is listed as 11mm on the built but links to 4mm on the Master Parts List & Link, above.  The standoff is 12mm and had to accommodate a screw from both ends, leaving 6mm for each screw plus the thickness of the board on either end, 3mm for the Electronics Board and 2mm for the Raspberry Pi.  So the screws should be no longer than 8mm.  I bought some m2.5 x 8mm screws at Ace Hardware & they work OK.  M2.5 x 6mm hex cap head to match the others would probably be better: https://www.mcmaster.com/92095a458

ericjunkins, 2018-09-27:
This issue will be resolved during the merge of pull request #16

vssystemluba, 2019-03-12:
Resolved by e243445

closed 2019-03-12 by vssystemluba