nasa-jpl/open-source-rover #41: Electronics Board holes for Serial Splitter too small; holes for 5V regulator misaligned.

JHPHELAN, 2018-09-17:
1) On the laser cut Electronics Board, in the lower right corner, the holes for the Serial Splitter PCB are too small.  They should accommodate the #4-40 screws needed for the #4-40 spacer, but they won't fit.
2) In the same corner, the holes for the 5V regulator should be **21mm** or 13//16" apart (instead of 20mm).  Also, according to both the diagram and the photo, the holes should be oriented on the opposite diagonal (NW-SE instead of NE-SW).  Won't know until we get to wiring whether that's important.

ericjunkins, 2018-09-18:
I do believe that might be correct about the 5V reg being flipped upside down. I will update the drawing to have the appropriate dimensions and be oriented correctly, sorry about that mistake. 

As far as moving forward for you, I think there are a couple of options.

1. Match Drill the hole pattern in the acrylic yourself. A 2-56 screw is relatively small and you would run little risk of cracking acrylic if you choose to do this and are careful.

2. Install the header pins upside down on the 5V regulator and then make sure to connect things accordingly.

3. Make your own layout and design for the electronics board assembly and let us know!

jaeme, 2018-09-18:
HI I can't seem to download the dxf files.
Can you please help.
Thanks

ericjunkins, 2018-09-18:
@jaeme What do you seem to be having trouble with? Do you have the repository cloned locally on your computer?

closed 2018-09-18 by ericjunkins

jaeme, 2018-09-29:
@ericjunkins when I go to the DXF file it comes up with the GCode - code.
What is the repository cloned please.
Thanks

ericjunkins, 2018-09-29:
@jaeme This link is for githubs documentation on how cloning the repository

https://help.github.com/articles/cloning-a-repository/

Do this on the main github repo at:

https://github.com/nasa-jpl/open-source-rover

The dxf files should be there for you to access

jaeme, 2018-09-29:
@ejunkins thank you for your help and apologies about being such a novice

commit 315c9bd0 (2018-09-18): Closes #41 by increasing hole size on serial splitter board to be 4-40 thru holes, updating the part used for the 5V regulator (previous part was incorrect part number), and updating corresponding mounting pattern. Made changes in the CAD assembly as well as updating the .DXF file for laser cutting
commit 34610a5c (2018-09-18): Closes #41 by increasing hole size on serial splitter board to be 4-40 thru holes, updating the part used for the 5V regulator (previous part was incorrect part number), and updating corresponding mounting pattern. Made changes in the CAD assembly as well as updating the .DXF file for laser cutting