nasa-jpl/open-source-rover #44: Can not access Mechanical/Mechanical Integration/Rover 3D view.pdf

CWRatliff, 2018-09-19:
All other 3d .pdf's are fine, but "Rover 3D view" shows only a small colored axis system at bottom left of page.

ericjunkins, 2018-09-20:
I tried a few times to generate the 3D pdf of the rover model, every time resulting in SolidWorks crashing. It might end up being too big to successfully  generate the 3D pdf from

ericjunkins, 2018-09-20:
I'm going to explore if there are other options to give a 3D model of the rover for someone to view who doesn't have access to any CAD software, but for now I'm going to remove the Rover 3D view pdf as it is blank.

closed 2018-09-20 by ericjunkins

CWRatliff, 2018-09-20:
Hi Eric,

Thanks for addressing the Rover 3D issue. I hoped to use it to resolve a potential engineering problem.

As I understand it, the middle wheel's motor mounting channel attaches to a 1.5" channel using two channel connectors. That's OK, but then the 1.5" channel attaches to the bogie using only one channel connector. This looks like an eventual metal fatigue problem.

I was unable to find a ServoCity part that looked like it would work.

I machined a .25 thick aluminum part that is nearly identical to the channel connector part in the flat view but has two tapped holes on the ends to allow connection between the 1.5" channel and the 3.75" bogie part.

I'm very willing to share this part design if anyone is interested.

Thanks, Wayne

Closed #44.

ericjunkins, 2018-09-20:
@CWRatliff Yeah I had looked for a while to find a part that interfaced correctly from ServoCity. 

You are correct about the middle wheel assembly attaching to the bottom of the rocker-bogie on only the one side. The channel connectors are steel however, and I don't think that metal fatigue would be your biggest downfall to this design, as the weight and stress that part sees are relatively low compared to the strength of steel.

What I do see happen is over time on a lot of the screws that go into the channel connectors that eventually they back themselves out from vibrations over time (even with loctite and lock washers it eventually happens). Now our rover sees a lot of moving, we take it to events all around, so has lots of car rides and transportation. This issue of screws backing out is worsened by the fact that there are only two screws on each piece and they are planar, not opposing, so as soon as one starts slipping out the other one will much faster as well. 

I'd love to see the part you designed for this, and as I said it takes very little load overall, it likely is a component that could be 3D printed if access to a machine shop/equipment isn't viable.

CWRatliff, 2018-09-21:
Hi Eric,

Attached is a picture and hand sketch of the part I described. Hope this is useful.

Wayne

ericjunkins, 2018-09-21:
@CWRatliff It looks like you replied to that via email, which it doesn't upload attachments from

CWRatliff, 2018-09-21:
[channel connector.pdf](https://github.com/nasa-jpl/open-source-rover/files/2406486/channel.connector.pdf)
[image]

CWRatliff, 2018-09-21:
[channel connector.pdf](https://github.com/nasa-jpl/open-source-rover/files/2406489/channel.connector.pdf)