nasa-jpl/open-source-rover #45: .dxf files for Electronics Board and Back body plate corrupted?

JHPHELAN, 2018-09-19:
Previously downloaded & laser cut successfully both pieces.  Recently noted post from member unable to download.  Tried myself - able to download both files, but got "unrecoverable error" in CorelDraw and "failed to insert dxf" in Fusion360.  Perhaps the file got corrupted?
Jim Phelan, Mars Rover project lead, Houston Robotics

09771400540, 2018-09-20:
Hi I want refund my premium subscription amount.. I don't want to use my premium. That to I don't like it without any password/OTP. Automatically money got deducted from my accoun

ericjunkins, 2018-09-20:
@JHPHELAN I just checked and was able to open both of them in e-drawings, as well as upload them to sculpteo and the .dxfs were all recognized normally. Can you try just re-downloading them maybe your local version somehow got corrupted

JHPHELAN, 2018-09-20:
Right-click and "save as" ?.dxf failed to open.  But this worked:
Click on file, click on RAW, Control-A to select all, Control-C to copy all.
Open a new text file in WordPad, Control-V to paste, save as "_filename_.dxf", close file.  
Open file with CorelDraw.  BINGO!  Save as "_filename_.cdr".
Don't know why simple "save target as" doesn't work!
Jim Phelan, Mars Rover project lead, **Houston Robotics**

ericjunkins, 2018-09-20:
As a question, do you have the entire repository cloned locally? To test this I ran a git-clone on the repository and just took the file from there and it worked fine. If you have it cloned locally then all you would have to do is run a 'git pull' to get the latest updated files and that should have all worked well.

JHPHELAN, 2018-09-21:
No, I  don't.  I should really do that.  I've git-cloned once on Raspbian, but never on Windows-10.  I'll look into that!
Jim Phelan, Mars Rover project lead, **Houston Robotics**

JHPHELAN, 2018-09-21:
Done!  Git for Windows install asks a lot of questions.  Defaulted everything except changed the text editor from Vim to Notepad++ which required another download (to which it conveniently linked but watch out for add-on downloads like WinZip updater).  Opened Windows Git GUI and selected clone existing repository.  Then cc the Git-hub Mars rover clone link to clipboard, pasted into the Windows git-clone source line.  The target line was confusing as I created a git-clone folder in the Mars rover project folder and it stopped saying "folder already exists".  So I just backed out to my project folder, deleted the git-clone folder and put "git-clone" as the target folder.  Git then created & populated the folder.  Clicked on the .dxf file and it popped open (well s-l-o-w-l-y blossomed) in CorelDraw.  Sweet!  Appreciate y'all taking my nit-picking Issues seriously and so quickly updating the project!  Houston Robotics is really excited about this project (naturally, this is Space City after all) and looking forward to further collaboration as we progress with the build.
Jim Phelan, Mars Rover project lead, **Houston Robotics**

ericjunkins, 2018-09-21:
Glad to hear it worked :)

closed 2018-09-21 by ericjunkins