nasa-jpl/open-source-rover #47: sculpteo rejecting DXF files for the body

davidadkins1, 2018-09-26:
I tried uploading "Back body plate.DXF" and "Electronics Board.DXF" to https://www.sculpteo.com/en/upload/ and they were both rejected with:

"The uploaded file seems to be corrupt or lack the geometric data we need to process it. Please try with another file..."

ericjunkins, 2018-09-26:
Can I ask how you got and then uploaded the DXF files? 

I have my local repo linked to the github page (up to date on the latest commit), and I took those DXF files and uploaded them one at a time to Sculpteo and they worked fine.

https://www.sculpteo.com/en/print/electronics-board-129/mGshpa2J?uuid=au6WgUFh4lD3PYAe2go8k
https://www.sculpteo.com/en/print/back-body-plate-164/mGS3pbnT?uuid=Xl159DK6KjO2JhngRabCse

Are you able to click on those? I think I made it so those are shareable from their site

davidadkins1, 2018-09-30:
In Chrome for Windows 10 I right-clicked 
https://github.com/nasa-jpl/open-source-rover/blob/master/Mechanical/Body%20Assembly/Laser%20Cut%20Parts/Back%20body%20plate.DXF
Then left-clicked "Save link as..."

Which save the .DXF to a file.  I see now that appears to be a mistake, but I could find a download button for the files.

When I click your links above it takes me here https://www.sculpteo.com/en/dashboard/ with no obvious access to the files.

davidadkins1, 2018-09-30:
I was able to save the files using RAW mode.  You can close this ticket except for one minor issue. The PDF directions say to set the thickness to 1/8, but the only option is 3mm.

jaeme, 2018-10-02:
@ericjunkins I was just wondering if it was possible to get the overall dimensions of both dxf files as when I have opened them they appear to be scaled down. Your help is greatly appreciated.
Thanks

ericjunkins, 2018-10-02:
@davidadkins1 There is 1/8" option for Acrylic, see below picture:

@jaeme I have screen shots of both with dimensions below. The electronics board should be (203.2mm x 279.4mm) or (8in x 11in) and the back panel should be (233.2mm x  118.9mm) or (9.18in x 4.68 in) 

[image]

[image]

davidadkins1, 2018-10-03:
@ericjunkins Thanks. Apparently I didn't notice the "EU/USA preferred" button. It has to be USA for 1/8 to be available. Hopefully 3mm will be close enough.

jaeme, 2018-10-03:
@ericjunkins thank you so much for supplying this info really appreciate it.

closed 2018-10-03 by ericjunkins