nasa-jpl/open-source-rover #48: T2 spacer/standoff - should it be 3/16 instead of 3/8?

JHPHELAN, 2018-09-29:
In the Rocker-Bogie Assembly the build parts list does not include **T2 the 6-32 3/8" standoff** that is mentioned in the build.  Two are shown attaching an F Pattern Bracket to each side of the 2nd modified 3" U channel with two B3 6-32 x 1/2" screws.  There is no mention of using a nut, but the 3D pdf shows the screws penetrating the F Pattern Bracket, standoffs, U channel, hex nuts and extending into the U channel.  (Washers are neither mentioned nor shown.)  Using a 1/2" screw as listed, the screw barely penetrates the F Pattern Bracket and the 3/8" spacer.  Trying a 1/4" spacer the screw penetrates the F Pattern Bracket, spacer and U channel but not sufficiently for a nut to grab on, let alone penetrate the nut and beyond.  Perhaps what was meant was a 3/16" standoff or spacer?  That should allow the nut to attach but not be penetrated.  A 3/8" or even 1/4" spacer seems to give more room than necessary between the F pattern bracket and the U channel.  The 3D pdf suggests the space is tighter.  **Was a 3/16" spacer intended instead?**  And/or was a longer screw than 1/2" intended (5/8 or 3/4")?  
Issue discovered by **Derrick & Alfredo**, Rocker-Bogie build team, Mars Rover project,  **Houston Robotics**

Yet another nit-pick from
Jim Phelan, Mars Rover project lead, **Houston Robotics**  ;-))

JHPHELAN, 2018-10-30:
We completed the rocker-bogie build and 3/16" inch spacers work well.
JHP Houston Robotics

JHPHELAN, 2018-12-05:
In Rocker-Bogie Build.doc, section 2.2 "...,standoffs T2,..." should read "T1".  The T2 3/8" standoffs are too long and other Issue notes suggest the 1/4" T1 are correct.  We used 3/16" standoffs and they worked ok too.  Recommend editing Rocker-Bogie Build.doc to reflect this.

Jim Phelan
Mars Rover project lead
**Houston Robotics**

JHPHELAN, 2018-12-05:
P.S. Searching the Master Parts List the "T2 3/8" standoff" calls for quantity of 4.  An advanced Acrobat  search of all the build documents finds T2 only in the above Rocker-Bogie Build.doc.  T2 should be dropped from the Master Parts List.

Jim Phelan
Mars Rover project lead
**Houston Robotics**

ericjunkins, 2018-12-05:
Hi @JHPHELAN 

Thanks for the feedback this. I have been seeing issues and comments you've been opening up as you work through the build, and i appreciate the comments. It would be of great use to me moving forward if you find things wrong to help me by submitting a pull request with your proposed changes, there is a big backlog of things to do for this already and I am quite busy with my projects at JPL as well, so any help with issues like these are greatly appreciated!

ericjunkins, 2018-12-14:
Closing based on commit e1d6c9d

closed 2018-12-14 by ericjunkins