nasa-jpl/open-source-rover #49: Github Enhancement Proposal

debpedrano, 2018-09-29:
Hello Project Owner, can I make a logo design for this project? I am a open source contributor and I am willing to work with you.

ericjunkins, 2018-09-29:
Hi, we'd love to hear some of your ideas! I think over email is a better way, feel free to send me a message at eric.junkins@jpl.nasa.gov

debpedrano, 2018-09-29:
okay, thank you for your response. I will message on your email after making a design.

debpedrano, 2018-10-01:
@ericjunkins 

how about this one?
[image]

debpedrano, 2018-10-01:
[image]

spockbuddha, 2018-10-01:
MAKE IT DYNAMIC  PLEASE

debpedrano, 2018-10-01:
@spockbuddha , what do you mean by dynamic? I am sorry, I am not professional so I need your further perspective or explanation about your desired design. Thank you

debpedrano, 2018-10-12:
Hello, can I update the issue? please response. Thank you :)

debpedrano, 2018-10-18:
@ericjunkins  Thank you :)

vssystemluba, 2019-06-13:
Closing this issue in favor of the more active discussion on the OSR forum at https://www.tapatalk.com/groups/jpl_opensource_rover/mission-logo-for-the-open-source-rover-t104.html.

closed 2019-06-13 by vssystemluba