nasa-jpl/open-source-rover #50: Logic Shifter board testing confusion - put DIR & OE to GND instead of GPIO

JHPHELAN, 2018-09-30:
In testing the Logic Shifter Board section 4 and Figure 22 indicate that the DIR and OE pins must be LOW (0V or GND).

DIR (pin 1, upper left) can be seen to connect on the schematic (via pathway "11" and demonstrated by following the PCB trace and proven by continuity check with a voltmeter) to header pin P4-3.

OE (pin 19, second right) connects (via pathway "10") to header pin P1-2.
These pins are shown connected by grey and white jumpers to Raspbery Pi GPIO pins 10 and 12.

**However, unless there is a script running on the Pi to cause these pins to go LOW, they will "float" and be neither HIGH nor LOW preventing the chip from functioning properly.  There is no mention of running a script on the pi for this test.  Instead, we connected them to two of many spare GND pins on the RPi such as physical pins 9 and 14 to force them LOW.**

+5V (P1-1) red wire to RPi +5V (pin 2).  Check.

GND (P4-4) black wire to RPi GND (pin 6).  Check.

During testing at each step the +3.3V purple wire to RPi +3.3V (pin 1) needs to march down the logic shifter headers P1-3 through P4-2.  At each step confirm the +3.3V on the corresponding Bx pin on the right side of the shifter (according to the schematic) and look for the +5V on its partner Ax on the left side one pin up.

Thanks to **Roberto Pensotti**, Electronics build & testing team, Mars Rover project, **Houston Robotics**

Jim Phelan, Mars Rover project lead & chief fault finder ;-) , **Houston Robotics**

ericjunkins, 2018-10-02:
@JHPHELAN It would be correct to either connect the DIR and OE pins to either a GPIO pin driven low (effectively grounds the pin) of straight to the GND pin. I guess one important distinction is that the above bolded statement should actually read "they CAN float". When I was testing the boards in the above manner I had both the OE and DIR not connected to anything, and in practice you shouldn't see any floating behavior, however it would be correct to connect them to GND or a GPIO driven low.

vssystemluba, 2019-07-26:
Closing as this is no longer relevant after the merging of the electronics-pcb-rework branch 1245e914998c893c12e6b350123e3883899f1534

closed 2019-07-26 by vssystemluba