nasa-jpl/open-source-rover #51: Voltage Divider Board testing reversed

JHPHELAN, 2018-09-30:
**Houston Robotics**, Mars Rover project,  Electronics build team member **Roberto Pensotti** beautifully populated & soldered the three PCBs.  While testing the Voltage Divider all checked out according to the directions up until step 9 at which point it failed.  Repeating the test together during our weekly MeetUp failed again as did a fresh attempt at home the next morning.  Instead of measuring just under 2V, it measured 0.2V or less.  IC chip placement, resistor values & soldering technique all passed inspection.

Tracing the directions back from the pin number to the header designation to the schematic to the encoder function revealed that +5V signal voltage was being applied to the ENCn OUT pin and measured at the ENCn IN pin which seemed reversed.  Reversing the directions, placing +5V at the ENCn IN pin & measuring at the ENCn OUT pin measured consistently about 1.6V.  Test passed!

Jim Phelan, Mars Rover project lead, **Houston Robotics**

ericjunkins, 2018-10-02:
@JHPHELAN Ah yes, you are correct, the pictures in the PCB testing document show putting the +5V signal on the output side of the pins now the input, this is what I get for not taking those pictures DURING my initial testing and only doing it afterwards.

 It would take me a bit of time to get a board back out of the robot and get it in a spot to be able to take those pictures again. Would you be willing to submit some pictures to correct this error? Showing where to put the +5V test signal and where to test? I'd greatly appreciate the assistance

ericjunkins, 2018-12-27:
@rcywongaa Thank you for taking the time to add a few pictures of this process for us. Do you feel like you know how to add those pictures to the documentation as it is now in the PCB testing doc? It would be useful to us if changes come in the form of pull requests that we can review instead of issues for us to fix entirely. I'm happy to help walking through that process if you need :)

In addition I'm currently working on something that I think upgrades the electronics system quite a bit. I designed a PCB that all the motor controllers, voltage regulator, and Raspberry Pi kind of just "slot" into and will handle a lot of the electrical wiring inside the robot. 

I just finished designing it and need a little bit of time to test it and make any design revs. Once that's done I think some of these documents will be changing to incorporate that board instead. As for another ask from the community the board is designed using a free PCB design software KiCAD so I can open source the entire board project, not just the board Gerber files.

rcywongaa, 2018-12-28:
Sounds good!  Any idea roughly when the new PCBs will be ready? A few weeks? months?

I'm roughly 40% through the electrical wiring, would gladly wait / redo / work on other parts if it means a cleaner build.  I suppose the PCB is only for the power, raspberry, motor controller connection? (i.e. I'll still need to do my own motor controller to encoder / motor wiring)

ericjunkins, 2018-12-28:
@rcywongaa I have the boards designed and ordered, I get them the first week in January, and will hopefully be able to test it within about a week or two. I still haven't come up with a way to mount them into the rest of the robot as well so that will take just a little bit of time as well.  Below are some screen caps of what the boards look like. 

## Motor_control_Board
Dimension is ~ 5 x 6.5 inches
[image]

[image]

You would have a change in the internal routing of wires where most components talk through the board, but everything outside of the main body would stay the same, with just a 6-pin terminal block per motor to hook up. So your wiring harness from the body to each of the motors would stay the same, might be a good place to focus time. 

[Monolithic Board.pdf](https://github.com/nasa-jpl/open-source-rover/files/2715379/Monolithic.Board.pdf)

## RPi_zero_hat
I also decided it might be cleaner to put a RPi zero in the head, and run a USB-> TTL serial cable from the RPi 3B to the zero in the head. This RPi zero has it's own PCB hat i designed to handle the logic shifting and then give a standard cable to connect between it and the LED matrix as per suggestion on Issue #59 

[image]

I can keep you updated on the status of these as I receive them, do testing, and make any revisions.

PR #78 (2018-12-28): Fix pin numbers in voltage divider board testing instructions
Fixes #51
closed 2018-12-28 by ericjunkins