nasa-jpl/open-source-rover #52: How to add solar power charging similar to real thing?

dublo7, 2018-10-03:
I am interested in building one of these but want it to be self charging. Can someone submit plans to implement this feature?

aktfire747, 2018-10-03:
Make that two of us that would like solar power.

Main question would be. Have any tests been done on max current load while operating?
From that we should be able to work out the amount of solar input required to charge the batteries.

ericjunkins, 2018-10-03:
@aktfire747 Max current pull on the system depends heavily on the way you are operating the robot, mainly involving obstacles. If you are driving on flat ground there is a relatively low current pull on the system, usually in the 1-1.5A region. However when you get to climbing over obstacles where motors are seeing significantly higher torque you can see numbers quite a bit higher.

-------------------------------------------------------------------------------------------------------------------

As this is an education project I think first what is necessary is to have a discussion in general about how to design a solar charging system, identify the necessary components, and talk about the parameters that must be considered when trying to create a solar charging system for this particular robot.

I'm happy to be a part of this discussion, but I'd first ask you to do a little bit of digging on your own and share some ideas of the above topics.

aktfire747, 2018-10-05:
Thought we might have had a post from @dublo7 by now.

Identify the necessary components.
1. Solar panels
2. Regulator
3. Wiring or various sizes.

Operating modes
1. Solar panels only charges the battery(s).
2. Solar panels supply power to the rover and battery(s) only used when solar power drops off.
3. Solar panels and battery work together to supply the rover.

ericjunkins, 2018-10-05:
I think there are some things we can still add to this list, and break out a few of those into more concrete items. I'm going to quickly overview all the things that should be considered when looking at designing the power system for a project. It might be that in order to add solar charging capability you might want to change some of the other electronics to better match, ie. swap to a different battery. 

### First Order Parameters 
#### Battery
 *  Voltage
 * Capacity
 * Discharge/Charge Rate
 * Battery Chemistry (Mostly for safety reasons)
#### Battery Management System
 * Over Voltage Protection
 * Under Voltage Protection
 * Short Circuit Protection / Over current
 * Cell Balancing (If using a more than single cell battery)
#### Solar Cell
 * Voltage per cell
 * Output Current per cell
 * Number of cells
#### Robot Operating Parameters
 * Voltage Levels necessary (currently only 5V and Battery Voltage)
 * Current Draw
   * Nominally
   * Max
 * Desired Operating time

Now in the current build of the robot quite a lot of this is taken care of already, as the battery has it's own battery protection circuitry, as well as a cell balancing system. That being said I think there is a little more to look into than just plug a solar panel into it to charge the battery, as there is nothing handling what current is going to be supplied. One thing to do might be to look at the way the battery charger that is in the parts list handles this.

As @aktfire747 mentioned you could have different ways your solar panel interfaces into the system. Generally in my experience it is designed such that the solar panel charges the battery, which can simultaneously be pulled from the robot. The only real difference between your 1. and 3. bullets are that in 1. you have nothing you're doing with extra current not consumed by the robot and in 3. the extra current gets dumped into charging the battery which is much more preferable. 2. Requires another monitoring system to do switching, and likely isn't worth the effort of implementation., as there would be little gain from doing that.

aktfire747, 2018-10-11:
Looks like bullet #3 is probaly the one to use then, since it would make best use of the panel output. Keep battery type with what is for the robot now.

I am thinking with the solar cells, if we had a system that output 12 volts to match the range of raw volatge from the battery, then either use another regulater to feed in to the robot, (to reduce load on the one already being used for the battery), or feed the solar output in to the one regulator with the possible risk of overloading it.

As for robot operating parameters, voltage would have to be kept to what is set for the robot now. The current draw would be variable depending on how the robot is being operated at the time.

Desired operating time would be user dependant. With me I am thinking of having mine powered on all the time once built and testing is completed with shutdown only for maintenance.

A thought bubble.
I have seen a few solar regulators that have been using an Arduino and have wondered if something like this could be used to regulate the solar output feed in to the battery. My thinking, it could be linked in to the Raspberry Pi to provide feedback on battery / solar output to the robot's controller.

https://www.instructables.com/id/ARDUINO-SOLAR-CHARGE-CONTROLLER-Version-20/

ericjunkins, 2018-10-11:
@aktfire747 There are a few things I want to make sure you are aware of here. First the voltage of the battery is not 12V, it is a range between like 11.7-16.7V, depending on the charge level. Second when you say another regulator, I want to make sure that you understand that you need more than a simple voltage regulator between the solar panel and battery. The link you put https://www.instructables.com/id/ARDUINO-SOLAR-CHARGE-CONTROLLER-Version-20/ actually goes into this a little bit about how one takes solar panel output wattage and converts it to what you need.

**I want to stress the importance of understanding how exactly battery charging works before attempting to implement this, as if done improperly it could potentially be dangerous.**

aktfire747, 2018-10-12:
Think I will put things on hold for awhile, (others are welcome to come in of course) to do a bit more research on solar charging systems. @ dublo7 can keep going as well.

vssystemluba, 2018-12-06:
Closing this issue for now.  Let's recreate this in a pull request or an additional module in the project once someone builds it.

closed 2018-12-06 by vssystemluba