nasa-jpl/open-source-rover #54: Electrical assembly instructions figure 30

CWRatliff, 2018-10-08:
In fig 30, it appears that motor 1 is attached to M2 A&B but uses encoder #1. And similarly for the other 3 corner motors. I would have expected motor 1 to be associated with M1 A&B and encoder 1. Is this a typo or intentional?

On the other hand, I am having problems associating encoder 1 with Motor 1, when Motor #1 is wired to M1A/B and encoder 1's sensor wire (from input pin 2 to output pin 14 of the voltage divider)  is wired to pin 7 of  roboclaw 4.

Also: the Encoder Signals Schematic rev B  is inconsistent with figures 29 & 30 of the elec assy instrs and the PCB with respect to numbering of the header groups.

ericjunkins, 2018-10-09:
@CWRatliff Thanks for the feedback, you are correct in the typo, Motor 1 should be on RoboClaw4 M1A/B, I'll fix that shortly.

Your second point, it appears that since this I documented this information there has been an update to the RoboClaw Firmware, which to my best assessment switches the ENC input on the RoboClaws. Meaning that what was previous EN1 is now EN2 and vice versa. I've verified this with every RoboClaw I have on the Latest firmware version and it held true on all them after updating to latest firmware. I will have to figure out how to proceed on this, as in my documentation it would lead to wiring encoders switched, but doesn't match how the RoboClaws should be wired based on how they are labeled and documented.  

Your last point I'm having a hard time following which exact things are inconsistent.

ericjunkins, 2018-10-10:
So I added a hopefully more clear diagram showing how to wire up the corner system through the Divider board and what should be connected to what. I'll be trying to update the other figures to match this style, but it will take me a little bit to get around to that :)

CWRatliff, 2018-10-10:
Hi Eric,

Thanks for the info on the encoder reversal. I must have retraced my wiring a hundred times, looking for an error.

Wayne

ericjunkins, 2018-10-10:
I verified it by connecting directly jumper wire from the motor to the EN1 +/- pins for AB, and saw signal on the M2 encoder channel using the Ion Motion GUI. you might want to just verify that is in fact what you are seeing as well just as a double check.

vssystemluba, 2018-12-06:
Closing this issue since we have a workaround.

closed 2018-12-06 by vssystemluba