nasa-jpl/open-source-rover #61: Electrical Build Errors

JHPHELAN, 2018-11-07:
Electrical Build Errors

Electrical Build, Page 22, blue text box, “In addition, Figure 19 shows how we routed
the wires through the rocker-bogie, and then from the rocker-bogie into the main
body.”  Figure 19 should read Figure 21.

Electrical Build, Page 23, table, “2x 24AWG & 4x 30AWG”.  No 24AWG wire is specified in the parts list, only 16, 20 and 30AWG.  Presume to use 20AWG in place of 24AWG in this build.

James Phelan
Mars rover project lead
**Houston Robotics**

vssystemluba, 2018-12-06:
Fixed the first reference number issue in 71ad24c.

I'll follow up with Eric and verify what size of wire we're using and then update those docs accordingly.

vssystemluba, 2018-12-06:
Confirmed that we use 20AWG and 30AWG on our rover, not 24AWG.  I've updated the build documents accordingly to match the parts list.

commit 7c6ed4bd (2018-12-06): Fixed incorrect reference to 24AWG wire in Electrical Build doc.  Closes #61.
commit 3ecbd568 (2018-12-07): Updated parts list to include correct links to 30AWG wire.  Closes #66 and resolves #61.
closed 2018-12-06 by vssystemluba