nasa-jpl/open-source-rover #62: Serial Test Error

patmohd, 2018-11-08:
We are trying to test files using Serial Test, when we insert the following
python /home/pi/osr/rover/serialTest.py we receive a error saying 

Traceback (most recent call last):
  File "/home/pi/osr/rover/serialTest.py", line 49, in <module>
    main()
  File "/home/pi/osr/rover/serialTest.py", line 43, in main
    clientThread = threading.Thread(target=mythread.SendData, args=())
AttributeError: Threads instance has no attribute 'SendData'

Can anyone provide a solution for this?

mangunillos, 2018-11-23:
That's a bad identation of the code because looks that should be a method and the interpret take it as a function because of this.

mangunillos, 2018-11-23:
But then another error jumps becaus the name socket is not defined

vssystemluba, 2018-12-05:
@patmohd did you clone the entire rover-code directory?  It looks like you may be missing dependencies.

vssystemluba, 2019-03-11:
Closing this because of inactivity and because we have had other builders use this script without issue.

closed 2019-03-11 by vssystemluba