nasa-jpl/open-source-rover #68: Unable to establish connection to RoboClaw motor Controller

patmohd, 2018-11-30:
I have reached the part where all the electronic connections have been made and all the steps from ‘Software Steps’ have been followed. But when I connect the raspberry pi with the rover - electronics, and run the command - sudo python main.py -s -x, I face the following error 
Traceback (most recent call last):
   main()
File “/home/pi/osr/rover/main.py”, line 24 in main
   rover = Rover(config.args.bt_flag,args.xbox_flag,args.unix_flags)
File “/home/pi/osr/rover/osr.py”, line 30, in __init__ 
   super(Robot, self).__init__(config)
File “/home/pi/osr/rover/robot.py”, line 18, in __init__ 
   super(Robot, self).__init__(config)
File “/home/pi/osr/rover/motor_controller.py”, line 38, in __init__ 
   raise Exception(“Unable to establish connection to RoboClaw motor controller”)
Exception: Unable to establish connection to RoboClaw motor controller

vssystemluba, 2018-12-05:
A few of the first things I'd check:
- Have you verified that your roboclaws are plugged in properly?
- Do all roboclaws have power?
- Is serial communication enabled on the raspberry pi?  (did you follow every step in the software build guide?)
- Can you connect your computer to the roboclaw directly via microUSB cable using IonStudio?

ericjunkins, 2018-12-07:
@patmohd I'm going to make a note that there should be a bit more feedback in that Error, and that we should instead of have report back which motor controllers it did not communicate to. You're more than welcome to help us with that if you wish, by wiring that code on your own and submitting a pull request on it

vssystemluba, 2019-06-13:
Closing due to inactivity.  In the future, connection issues between the Raspberry Pi and the RoboClaws should be greatly minimized by the monolithic PCB board described in the `electronics-pcb-rework` feature branch.

closed 2019-06-13 by vssystemluba