nasa-jpl/open-source-rover #69: How many of us rover builders are out there?

JHPHELAN, 2018-11-30:


JHPHELAN, 2018-11-30:
How many of us are out there?  I can see from some of the posts that there's a least a handful.  Does NASA/JPL have any idea how many people/teams are working on this?  Who are they?  Where are they? What are they? Why are they doing it?  I'd love it if each of us would COMMENT on this with a little biography.  I'll start below.
Jim Phelan
Mars rover project lead
**Houston Robotics**

JHPHELAN, 2018-11-30:
**Houston Robotics**
Houston, Tx
Started a few years ago at TXRX Maker Space in Houston as Houston Robotics Club on Meetup.
An off-shoot "Artificial Intelligence & Machine Learning for Robotics and IoT" was later created.
As we grew, a couple of members associated with Houston Community College got us a larger classroom venue and access to the Fab Lab at HCC Workforce campus in Stafford, Tx while still making use of TXRX.  The two branches recombined in a 501c3 as Houston Robotics.  Our members have a wide variety of ages, interests and expertise from school kids to seniors, curious beginners to robotics professionals.  We have people with interest and expertise in mechanics, electronics, programming, AI, autonomous driving, drones, IoT, 3D printing, laser cutting, fabrication, etc.  If someone runs into a problem, someone else can usually help.  I think most robotics fans have admired and followed the Mars rovers since they were first announced.  When NASA/JPL first announced the Open Source Rover months ago, several of our members brought it up.  As a group we pounced on it as a project with something for everyone.  Right now we're finishing the wiring and continuing with testing and calibration.  We hope to have a fully working rover under our Christmas tree!  Because we have tried (unsuccessfully as we just can't help ourselves) to be as faithful as possible to the original NASA/JPL plans, we call our first version "Fidelity".  Once completed, we will put the Houston Robotics touch on it by adding sensors, cameras, LIDAR, and machine learning to make it an autonomous driving all-terrain vehicle named "Audacity".
Jim Phelan, project lead
**Houston Robotics**

xtream1101, 2018-11-30:
Hello Everyone,

I am building this just for fun as a one man team. I enjoy building and creating things of all kinds. I have always enjoyed robotics and do it as a hobby as well as creating electronics from scratch. 

I have finished building and wiring the rover, I also took the time to paint my metals matte black. Next step is to get the raspberry pi setup and connected to an android phone for the controls. Once I get it all working I think my next step will be controlling it from a longer range and have a camera on it. I have been interested in learning how to do autonomous driving with a few other rovers I have too but I don't currently have the time to dedicate to it right now.

Eddy Hintze

kkohlsaat, 2018-12-01:
Me and a team of 5 members are building this project as our senior project. Updates and pictures are soon to come! We're in the process of buying parts as well as printing as much as we can to reduce cost. Looking into calling our rover POL-EE, name not final.

Karl Kohlsaat
Cal Poly San Luis Obispo

JHPHELAN, 2018-12-01:
Karl,
Thanks for posting your story.  
Love the name!
Although the JPL team did a GREAT job on the BOM, there were a few mistakes.  Before you buy, read the Issues and check that the web link is to the right part.  We found the wire braid too short (we would have needed more) and cumbersome to stuff.  The very fine wire was easy to break and awkward to crimp into DuPont connectors.  We ended up using CAT5/6 cable & using 1 pair each for + and - power a la Power Over Ethernet and the remaining 4 wires for the encoders.  Preliminary testing is successful.  We'll do a full post eventually.  

Jim Phelan, Mars rover project lead
**Houston Robotics**

xtream1101, 2018-12-01:
I am in NYC. I did not go the route of anodizing the parts, though it did cross my mind. I just ended up using spray paint (metal primer, black, and a clear matte coat). Its holding up pretty well so far.
That 30awg wire is _very_ thin, I was so worried I was going to break it too while running the wires. I ended up soldering them to other wires with dupoint connectors already on them.

I don't have a name for my rover yet, its always the hardest part for me.

My photos so far: https://photos.app.goo.gl/ZK7aEih7C9tU4wB1A

JHPHELAN, 2018-12-01:
Nice wiring job!  Badass paint job!  Sweet!  Nightstalker Ninjabot!

JHP

aktfire747, 2018-12-01:
Not started the building yet, but hopefully next year. Also hoping to add solar power to the rover as well.

vssystemluba, 2018-12-05:
All,

I love the enthusiasm here and that we get to meet so many of you wonderful people who are building the rover!

That said, I'd like to suggest that we take this discussion over to the JPL Open Source Rover forum at https://www.tapatalk.com/groups/jpl_opensource_rover/index.php. (We even have an open ["hello" thread](https://www.tapatalk.com/groups/jpl_opensource_rover/hello-t4.html) that's specifically dedicated for introducing yourself!  We also have a [section where you can show off your completed rovers](https://www.tapatalk.com/groups/jpl_opensource_rover/completed-builds-f7/) and all the fancy things you'll add to them.) The forum is better suited than GitHub to show images and facilitate a lively discussion, and that way we don't have a discussion represented as an open "issue" on the repository.

I'll close this thread, but highly encourage everyone here to duplicate your comments above and then continue the discussion over on the forum!

@JHPHELAN @xtream1101 @kkohlsaat @aktfire747 @wmburke ☝

closed 2018-12-05 by vssystemluba