nasa-jpl/open-source-rover #70: I can't find the absolute encoders on US Digital site

weigu1, 2018-12-03:
Hi,

I can't find the absolute encoders on the US Digital site (E7 1/8 inch, 10 Bit Analog, Sleeve Bushing (N)). There is also no link in the bom. Is there a replacement type?

Thanks

xtream1101, 2018-12-03:
The link is in the Master Parts list sheet: https://www.usdigital.com/products/encoders/absolute/rotary/shaft/MA3

Make the selections in the **Product Configuration** column on the right

spockbuddha, 2018-12-03:
I am too rational......i presume

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/nasa-jpl/open-source-rover/issues/70#issuecomment-443758100",
"url": "https://github.com/nasa-jpl/open-source-rover/issues/70#issuecomment-443758100",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [nasa-jpl/open-source-rover] I can't find the absolute encoders on US Digital site (#70)",
"sections": [
{
"text": "",
"activityTitle": "**xtream1101**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@xtream1101",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"nasa-jpl/open-source-rover\",\n\"issueId\": 70,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"nasa-jpl/open-source-rover\",\n\"issueId\": 70\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/nasa-jpl/open-source-rover/issues/70#issuecomment-443758100"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 419115765\n}"
}
],
"themeColor": "26292E"
}
]