nasa-jpl/open-source-rover #74: Calibration update & issue

JHPHELAN, 2018-12-10:
Calibration update & issue

The Calibration document refers to a program Ion Studio at 
http://www.ionmc.com/downloads.
This link redirects to
http://www.basicmicro.com/downloads
and the program is now called BasicMicro Motion Studio found under RoboClaw General Downloads.

Aside from that the program looks the same.  Installation and General Settings no problem EXCEPT the Logic battery would read 2.7V instead of 4.5V.  Otherwise it seemed to run ok.  We changed the General Settings alerts to permit 2.7V.  We've yet to figure out that discrepancy.

PWM settings were no problem and indicated the occasional need to swap connections.  We were careful to zero the encoder values before starting each trial.

Under _Velocity Settings_ at max speed the M1/M2 Speed boxes showed rapidly changing values.  Sometimes in a fairly narrow range of, say, 7,000-8,000 where we could estimate an average of 7,500.  The other motor, however, may range from 27,000-29,000 which seemed wildly different!  

Entering the estimated max speed into the QPPS box and clicking Tune M1 causes the motor to slowly rotate back and forth a couple times as mentioned.  It then continues to rotate in the same direction forever.  No values are populated in the P, I, D boxes.  Motion Studio has to be stopped via Task Manager while the motor continues to run at its last commanded speed until power is removed by the kill switch.  Trouble shooting is still in progress, but if anyone else has insight into this problem it would be appreciated.

Will also post to the Forum

Jim Phelan
Mars rover project lead
**Houston Robotics**

basile-laderchi, 2018-12-10:
After a little digging around in the [Wayback Machine](https://web.archive.org/) I found the attached setup file that says it's going to install the Ion Studio program that is referenced in the documentation.
I hope this is of some help.

[setup.zip](https://github.com/nasa-jpl/open-source-rover/files/2662338/setup.zip)

Basile Laderchi [:|]

JHPHELAN, 2018-12-14:
Solved (partly)!
The issue was due to the low Logic Battery Voltage.  Appears the connector from the battery to the 5V regulator, and thus to the RoboClaw Logic Battery input was disconnected.  So why did it not show zero voltage?  Suspect the RoboClaw was getting some logic voltage via the USB cable from the computer.  Once the 5V supply was restored, the calibration proceeded (mostly) as described.  We haven't solved all the calibration issues so not ready to Close, but this is a major step.

Jim Phelan
Mars rover project lead
**Houston Robotics**

ericjunkins, 2018-12-14:
@JHPHELAN Good job so far. Let me start by answering your most recent question about logic battery voltage. There is an internal voltage regulator on the Roboclaw motor controllers that takes power from the Main battery (+ and - terminals where you power the board) and regulates it down to a certain voltage (off the top of my head I think its in the 2V range). This is sufficient for a number of standard quadrature encoders, for instance the quadrature encoders on the pololu motors will run on this voltage. You only need to run 5V to the logic battery for the purpose of powering the absolute encoders which only run on 5V.

Earlier you mentioned while doing calibration the wheels would spin forever and the program would freeze, this is likely because of an issue in direction of the motor vs encoder. If you send positive PWM to the motor it expects that the encoder values strictly increase, if it does not then the auto-calibration process fails. 

There are ways to tune these numbers yourself, you can learn a little more about it here:

https://www.crossco.com/blog/basics-tuning-pid-loops
https://robotics.stackexchange.com/questions/167/what-are-good-strategies-for-tuning-pid-loops

I will update the docs to point to the newly updated Basic Motion studio as well thanks for the info on that.

vssystemluba, 2019-06-13:
Based on the discussion above and an excellent writeup on the OSR forum at https://www.tapatalk.com/groups/jpl_opensource_rover/viewtopic.php?f=10&t=68&p=352&hilit=calibration#p352, I will mark this issue closed.  Thanks everyone for your input and helping get this solved!

closed 2019-06-13 by vssystemluba