nasa-jpl/open-source-rover #91: REV-C Revisions to be made

ericjunkins, 2019-02-21:
I've received the REV-C board, and these are the following revisions to be made:

  * All coupling capacitors are connected to the voltage side of powering components instead of GND. They will all need to be fixed for this.
  * Need to add a USB connector on 5V line to run to the 4 position powered USB hub (optional)
  * Need to add a Barrel Jack connector to board for 12V powered USB hub (optional)
    * https://www.digikey.com/product-detail/en/tensility-international-corp/10-01776/839-1242-ND/5638252
    * https://www.digikey.com/product-detail/en/tensility-international-corp/10-02382/839-1312-ND/6593491
  * 12 V regulator mounting holes are 90 deg offset from pins, need to update mounting footprint
 

Arduino Sheild Hat Rev to make:
  * Move location of 5V terminal block to not be ontop of  USB connector on Arduino UNO
  * Standoffs need to be 2-3 mm longer

philwil, 2019-03-04:
I built most of one of these over the weekend 3/2/2019. I got 5 PCb's from JPL. 
They also sent me 10 of the arduino shields !! 
I only plan to use two of each of them so the others are up for grabs !!
I have yet to start testing since some components were out of stock.
Are there any plans for a REV-C.1 or a REV-D with the capacitor problem fixed ?
 If not I'll start building my second board.
I am also preparing a list of suggestions to add to Eric's list above.
I can focus on completing that list if you are trying to get the next rev out.
Thanks for the effort. This is a fun project.

ericjunkins, 2019-03-04:
I have a rev-d board ready to be populated, i'm just waiting on a few components from digikey still. If you want to test yours you'll have to replace all the caps with jumpers instead. 

The arduino hat board works but I realized the screw terminal pins run into the USB port and would short, i clipped the pins on the bottom and put capton tape over.  I have a 2nd rev of that as well. 

I will post the updated files as soon as I get to testing the new rev d board,  hopefully in the next few days

philwil, 2019-03-04:
Thanks, I'll hold off building the second board until I get your reworked files.

philwil, 2019-03-05:
HI Eric,
Some more feedback.
Got the two power supplies for the REV-C board last night.
The 5V PSU  fits OK but I can only use one spacer The other hits the
connector. I can use an extended bolt to hold it down.
The 12V PSU seems to have the mounting holes reversed.
I assume that both of these power supplies fit under the board.

I am planning on using much cheaper ten turn pots for the absolute encoders
I can get the larger bore pinion gears but am unable to use the grub screw
to secure them.
Looks like I'll have to glue the pinions on the pot shaft or shave a flat
on the shaft to give the grub screw enough thread to hold.
It looks like full range on the steering motors will use about 4.5 turns.
This almost gives us 2V if put 5V across the pots
I'll look at adding variable gain and offset to the op amps to get the 2V
range needed.

...

philwil, 2019-03-05:
> Thanks for the feedback, I had already 12v reg mounting pattern as the
> last bullet in this issue so that's fixed already.
>
> I'll let you know what I think is the best solution for the corners and
> you can let me know what your thoughts are on trying to implement it. I
> think it's a great idea to use a much cheaper positioning solution such as
> a potentiometer, but it would be best to do closed loop control from the
> encoder on the motor (no backlash, better resolution means better pid
> response). For this I see one of two options as really good solutions:
>
>    - read in the potentiometer straight to the raspberry pi, use this
>    data on boot to localize the corners position (maybe occasionally check and
>    update to see if slip has occurred)
>    - pass both potentiometer and relative encoder to the roboclaws, and
>    use a relay or mosfet to switch between which are being read, use software
>    to switch roboclaw read mode. The relay/mosfet can be controlled through a
>    gpio pin from the pi.
>
> Both of these allow you to change out the expensive encoders and save a
> few hundred dollars to the project, I'd really love to find that savings If
> you want to help design/ test this

On the encoders

I would much rather have the option to use the relative encoders on the
motors. ( I have them fitted) and then use the potentiometers to get some
form of absolute position.
Possibly use a MCP3008 for PI analog in .. dont populate it if you dont
need it. You would also get 4 extra spare analog inputs.
Another option would be to use a level detector on the potentiometer to
trigger detecting an absolute location.
If you use the potentiomenter you will have to solve the alignment problem.
You could use a trim pot on the board.
I like that idea since it is easy to set up.
Then we could use the relative encoders together with the potentiometer to
get the steering setup.
You bet I am willing to help here.

...

ericjunkins, 2019-03-26:
@philwil Sorry for the long delay on messaging back. I have though about that exact thing, using quadrature encoders on the corner motor and a potentiometer to sync the position and recheck occasionally if there is slip in the corner system. I'll have to think about the best way to build that into the board to support both, not necessarily an easy task at the moment. 

All revisions have been made and tested with commit 30e20808a101ab2fc350cca8feb0b62c04a7547c so im going to close this issue, feel free to continue the discussion though

closed 2019-03-26 by ericjunkins

philwil, 2019-03-26:
Thanks for the update.
I am busy on another small project at the moment but was just about to
place another pcb order.
I'll look at manually wiring the corner motor encoders at this time.
I also am thinking of an ADS1115 encoder to pick up the potentiometer
readings.
Using those for another project.
  I'll let you know when I order the new boards.