nasa-jpl/open-source-rover #94: Wire encoders for the steering motors

philwil, 2019-02-26:
I would like to suggest adding the extra encoder wires for the corner steering motor.
This will allow both the absolute encoder to be used or the motor mounted encoder.
In the latter case I am going to try a cheaper 10 turn potentiometer to derive a reference point and then use the motor encoders for steering position.

ericjunkins, 2019-03-26:
I'm trying to think of a nice way to integrate this in, without forcing users to go down this route. My initial though is having a switch on the board to toggle between absolute and relative encoders. It isn't exactly that straight forward to do but could be done that way. If you have any elegant solution that you think would work well feel free to suggest something

philwil, 2019-03-26:
Sure  try this
Wire in the extra encoder to each motor. That will not affect anything if
it is not used.

Then add  extra 3 pin connectors for the potentiomenters.
Take those wipers  to an optional ads1115 board (that is a plug in
addition). From what I see , if you add a series resistor on the analog
inputs the ADS1115 is protected against over voltage.

The main PI will have to reference the potentiomenter to the encoder if it
is  being used
The roboclaw will read the encoder but under position control from the
encoder not the analog input.

We'll need the extra 4 connectors for the pots but everything else should
be fine.

ericjunkins, 2019-03-26:
This doesn't deal with the fact that currently the ENA encoder line is run through voltage divider and filter before the roboclaw, which would destroy the data for the quadrature encoder

ericjunkins, 2019-03-26:
@philwil The thing that I'm concerned will happen is that even with the first 2 feature requests for the PCB it is segmenting it into configurations based on individual use cases/hardware that not everyone is buying. I want to avoid having this board that requires a specific configuration of solder bridges/switches/jumper shorts in order to work. 

With that I think a good path forward is to instead make an edit for the PCBs to accomplish these individual features. We can have a new branch of the repo for each of these edits, and change the PCB according to the feature. There can be a documentation edit for each of these, and then we can on the main README call out specific features / changes so that everyone is aware that they exist and how to build the project with them. 

As a question for you @philwil , do you have experience doing PCB design? I have a little bit of cleanup to do on the files before I post my KiCAD source files for the PCBs, but at that point are these changes you think you know how to make? Or are you looking for assistance in making those?

philwil, 2019-03-26:
Hi Eric,
I suppose you could remove R3, R5, R7 and R9 to isolate the input but I do
understand your concern.

I tend to do small PCB designs ( just started again ) currently only using
easyEDA online
If you can let me have access to the kicad files I think I can import them
to the online tool.

Phil

JHPHELAN, 2019-03-26:
Our team has the capability to modify PCB designs and there are a number of Houston resources to help if needed.  Appreciate your concern re getting over customized.
Jim Phelan
Mars rover project lead
Houston Robotics

ericjunkins, 2019-03-26:
Okay, I will try to  make it a priority to get the KiCAD source files cleaned up and uploaded so you can begin your edits. At this point I have everything except the E-STOP control from the RPi tested (might be able to get to that this weekend), but other than that the board fully checks out. When i upload the PCB source files I'll also push the latest parts list for the boards.

philwil, 2019-03-26:
Eric
Thanks That would be great
I have setup my pc for kicad.

vssystemluba, 2019-06-20:
We're going to close this as a won't fix due to the complexity of this feature request and our other priorities right now.

If it proves valuable enough to add in the future, we recommend that community members work together on it to outline requirements and edit the PCB.

closed 2019-06-20 by vssystemluba