nasa-jpl/open-source-rover #95: Enhancement - mini rc version

johndpope, 2019-02-27:
With so many eye balls on this project - it would be cool if someone could pick this up with effort to scale down project to size of a remote control car. Would make a neat smaller scale project.

johndpope, 2019-02-27:
[image]
http://www.moc-pages.com/moc.php/422507

rcywongaa, 2019-03-05:
Checkout https://github.com/Roger-random/Sawppy_Rover

vssystemluba, 2019-06-13:
While I love the discussion, it's not really an 'issue' against this repository, so I'm going to go ahead and close it.  I highly encourage you to continue the discussion above over on our forum at https://www.tapatalk.com/groups/jpl_opensource_rover/.

closed 2019-06-13 by vssystemluba