nasa-jpl/open-source-rover #103: Kill Switch option on new monolithic PCB

vssystemluba, 2019-03-20:
Per a DM I just received on the forum from @jhphelan:

Place a 4-position screw terminal strip on the board between main power output and the RoboClaw power inputs with heavy gauge jumper wires from + to + and gnd to gnd.  That way a kill switch cold be wired in replacing the jumper wires.  This would also allow a take-off for a "robot safety module" of flashing light & beeper to run only when the RoboClaws are powered.  This would permit power to the Raspberry Pi and Display for program testing while paralyzing the robot for safety.
Thanks for making this GREAT project!
Jim Phelan
Mars rover project lead
Houston Robotics

vssystemluba, 2019-03-20:
@ericjunkins I'll assign this to you for now.  If you've already implemented something like this, feel free to close.  If not, it seems like a good addition.  

Let me know what you think.

ericjunkins, 2019-03-20:
@JHPHELAN There is an ESTOP pin on the Roboclaws that disables the H-bridges on the motor controllers, disabling movement of the robot. In the current PCB Design I have that connected to a GPIO pin on the RPi, such that the PI must let the motor controllers know things are in a safe state. Are you specifically wanting an additional hardware E-STOP switch/button?

JHPHELAN, 2019-03-20:
Yes, using a keychain remote & a receiver relay an interrupt could be placed in the power train.  The ESTOP requires the RPi to be functioning perfectly.  The risk is that a program glitch could make the RPi "go off into the weeds" taking the rover with it.  A remote kill switch would allow the "robot safety officer" to stop the rover should it go rogue.
Jim Phelan
Mars rover project lead
Houston Robotics

ericjunkins, 2019-03-21:
@JHPHELAN Okay I understand now what are you asking for. I have to look at a few things.
The Roboclaw might support multiple E-STOP inputs. I know you can configure S3 to be an E-STOP input, you might be able to also configure S4/S5. If you cannot then I think I will add the following AND logic gate comparitor to the system. 

https://www.digikey.com/product-detail/en/texas-instruments/SN74HC11N/296-8217-5-ND/376871

As a few notes:

The signal from the RPi for the kill switch is set such that initially the motor controllers will be in an E-STOP mode, and require the logic line driven HIGH in order to clear that, which must be held HIGH at all times in order for the E-STOP error flag to be cleared and the H-bridges enabled. I was going to write a separate watcher on the Pi that would continually monitor the system and if anything happens it will drive that to LOW. 

This does mean that if there is a malfunction on the Pi catastrophically that you could get into the scenario you are describing. However I would suggest not putting the "kill switch" on the power line to the RoboClaws, and additionally definitely do not make it disconnect both power and GND. When the E-STOP flag is called on the motor controllers they will disable the H-bridge and it will safely handle all of the energy currently in the system. However if you cut power abruptly there is a chance there is still energy in that system, which then has potentially no safe outlet to go to. Also if you remove the GND connection you allow the voltage at the device to be floating, which could lead to some large voltage differential between the power terminals and battery upon connecting (especially if the system is just hot-plugged off). There should be no reason to ever disconnect the GND of the devices, only every the (+) side voltages.

All of the above mentioned are of course low chances of things that could go wrong, but especially in an educational project it is better to err of the side of caution

JHPHELAN, 2019-03-21:
Eric,

The kill switch should not, for the reasons you say, interrupt the 
ground.  But it does need to tap into it for the receiver circuit to 
operate.

The power interrupt seems simpler and has the advantage of the "robot 
safety module" takeoff option.

However, using the E-STOP, the kill switch relay could simply ground the 
E-STOP line from the RPi preventing it from activating the RCs without 
the need for a comparitor.

Can you make the RPi drop the E-STOP enable if it looses radio contact?  
I don't know the details, but Bluetooth does seem to be aware when it 
looses contact.

As far as an onboard switch, a big red "bop to stop" switch on top 
wouldn't be a bad idea!  I looked at several emergency stop switches but 
they were all expensive.

We did change our power toggle to a lighted push-button so that it would 
be very clear when the power was on.  We're also thinking of LED 
lighting for the inside to show it off through the acrylic back, and 
additionally front, panels (thickened to 1/4" for strength).

Jim

James H Phelan
"Nihil est sine ratione cur potius sit quam non sit"
Liebniz

ericjunkins, 2019-03-21:
@JHPHELAN  can you send me a link to the receiver you are using so I know how it works? 

  * The RPi can monitor if there is connection from a wireless device yes, there are a few ways to do this and make that descision
  * I had thought about installing power LEDs as well, I didn't because the voltage monitor will turn on and give indication of what is happening. In addition with the new PCB rework and adding an Arduino you can get some LED indication from the face that things are running instantly. I have a lot of code changes to upload when i get minute on that still.

ericjunkins, 2019-03-26:
@JHPHELAN I want to reference you to a comment I just made in #94 for future PCB design requests. Would you mind reading that and let me know if that path forward is something that works for you, and also let me know if you are able to make board edits yourself or if you are looking for assistance in designing them.

ericjunkins, 2019-04-05:
@JHPHELAN I verified today that you can configure the RoboClaws to accept multiple E-STOP signals onto different control pins. My suggestion to the remote kill-switch edit you are wanting to add is to add that to S4 or S5 of each of your Roboclaws on the PCB

vssystemluba, 2019-06-20:
JPL is focusing on issues other than this one, so we're choosing not to implement it on our end.

If it's a valuable enough feature for other folks, you're encouraged to try your hand at adding it based on the discussion above.

closed 2019-07-26 by vssystemluba