nasa-jpl/open-source-rover #106: Creating a "build packet" and release pipeline

kevincloutier, 2019-04-12:
Per the conversation over in the [forums](https://www.tapatalk.com/groups/jpl_opensource_rover/what-to-do-when-we-find-issues-t87.html), let's consider this feature request to help reduce obstacles and bring down the barrier to entry for the project.

> We want to make it as easy as possible for people to get involved. The structure was designed with modularity in mind, and being able to have many different builds of subsystems in manageable ways. **_But it might be smart to consider implementing a more "build packet" version for those looking for instruction kit with less navigation. It is definitely worth thinking about._**  
- emphasis mine

Achllle, 2020-06-25:
We love this idea. If anyone has any idea how to get this done, we'd love to hear it!

Achllle, 2023-09-25:
Years later, we're finally making progress on this with the release of v2 and Gobilda's OSR order page. While it's not a complete build packet, the number of manufacturers to source parts from has drastically gone down to the point that this no longer presents as large of an issue. While the build packet isn't there yet, since it's on its way and to clean up some old issues, I'll close this.

closed 2023-09-25 by Achllle

kevincloutier, 2023-09-25:
Hi Achlle, great news... BUT, I think the intent of this issue may have been lost over the years. This was originally created to address the lack of a "build packet", that is, provide students and makers a packet of information that contained everything they needed to get going (e.g. not have to build LaTex, etc. but rather have a simple book-like document ready to go). It's been years since I have been following the project, so maybe this is no longer an issue... but the ticket wasn't related to sourcing parts etc. Thanks for working on the project though, I had a blast participating way back in 2019. I hope all is still going well and there are hundreds, no, thousands, of Rovers driving around high school gymnasiums.

Achllle, 2023-09-25:
Gotcha - we did switch to readme files instead of latex a few years back and recently we created a website that has all the documentation autogenerated from the readme.md files, so I think that intent is accomplished. Most of the work going forward would be improving and extending the documentation for this new build version. 

Maybe the time has come for you to start building again? We could use some experienced people like yourself 🙂

kevincloutier, 2023-09-25:
That would do it, so it looks like this issue is def. closed! Nice work! The LaTex was super interesting to use and learn, but def. a barrier to entry for most.

If you seriously need some help, I would be interested in learning the state of the project to see if my expertise could be utilized. Maybe I can lend a hand or two... shoot me a message, email, etc. I always wanted to add a LIDAR to the Rover, maybe you guys already did...