nasa-jpl/open-source-rover #111: Aluminium and PVC tube stock

lukeweston, 2019-05-04:
Just a quick question:

The 5 feet of specified PVC tube appears to be only used to cut a single 6” length to mount the “head” assembly, and nothing else?. Is that correct? (I recognise that the mast may be a little taller, for sensor or camera use, but it seems like a foot or two of stock material will be more than adequate. Since McMaster-Carr is inaccessible to non-USA customers, I need to make some adjustments to the BOM.

Similarly, the 12.7mm aluminium tube used in the differential pivot assembly is cut into a 13.5” piece and a 15” piece, but there appears to be only one piece used? Where does the 15” piece get used, I can’t find it? I will source this material in a single piece without buying 3 feet of it.

Similarly, the 1/2” tube in the differential pivot assembly appears to be clamped on the outside and therefore its ID (and hence wall thickness) is not critical and may be changed. Is this correct?

ericjunkins, 2019-05-04:
You do only need the 6inch segement for the head assembly for PVC.

The 0.5 inch (12.7mm) aluminum tubing is used in the differential pivot for the 13.5 inch piece, and the 15 inch piece is used in the mechanical integration step. We are looking into solutions that wouldn't require buying these as one piece, and instead getting two pieces of exact length ordered.

Wall thickness does not matter, as long as it is still sufficiently thick to resist bending.

ericjunkins, 2019-05-04:
@lukeweston Do you know about the forum we have? 

https://www.tapatalk.com/groups/jpl_opensource_rover/

This is a more appropriate place to ask questions, as these are not really issues that need fixing. General questions and discussions are better suited for the forum

closed 2019-05-04 by ericjunkins