nasa-jpl/open-source-rover #113: Heat-set-inserts: which part #?

JHPHELAN, 2019-05-14:
The link in the .pdf for optional CAD 3D printer parts refers to heat-set-inserts and refers to:
[https://www.mcmaster.com/heat-set-inserts](url)
but it's nonspecific.  I ASSUME the reference to #6 means 6-32.  
I also assume straight instead of tapered to avoid additional machining.
I further assume the shorter version as the ledge in the new head for the Arduino isn't very thick.
Therefore I believe this is the appropriate part:
[https://www.mcmaster.com/94459A280](url)
I'll probably buy some 4-40s also for further use and maybe some long ones to make the postage worthwhile.  BUT PLEASE TELL ME if I'm way off before I order!  
Thanks!
Jim Phelan
Mars rover project lead
USAi Labs (formerly Houston Robotics)

ericjunkins, 2019-05-14:
@JHPHELAN Thanks for bringing this up, i'll add clarification on that. You do want the taper inserts, the 3d printed pieces all have taper holes and the acrylic will melt accordingly and will be fine to use the taper ones as well. You do want the short versions of all them. Here is the link to the #6-32 insert you would want: https://www.mcmaster.com/93365a130

JHPHELAN, 2019-05-14:
Ah, thanks!  Order going in now.

closed 2019-05-14 by JHPHELAN