nasa-jpl/open-source-rover #115: Issue Raspberry Pi and Controller

PhilippHaid, 2019-05-16:
Hello to all,
I'm part of a project, where our task is to build the rover. 
Our problem is that we have several problems with the Raspberry Pi and the controller.

While trying to run the codes and Programms on the Raspberry Pi to Control the rover by using the Xbox controller we got an error message.
This messages is:
When trying to run the main.py we got the massage : "Exceptation: Unable to establish Connection to Roboclaw Motor controllers".

thanks for any answers!

Philipp

vssystemluba, 2019-05-16:
Hi @PhilippHaid!

We've seen that error message pop up in the following 2 cases:
1) The RoboClaws are disconnected
2) There is a wiring issue or accidental wire swap between the Raspberry Pi and the RoboClaws
3) Serial communications have not been enabled on the Raspberry Pi

Have you verified that your wiring is correct between the Raspberry Pi and the RoboClaws, and are you able to run the `serialTest.py` script successfully?

ericjunkins, 2019-05-16:
@PhilippHaid  You also have to make sure each of the roboclaws have "multi-unit mode" enabled, this allows the parallel serial communication to each of them. If any single one is missing it the entire set will be unreachable.

In general this is better suited for our forum: https://www.tapatalk.com/groups/jpl_opensource_rover/
Under the Help and troubleshooting section. Issues on github are more for things that need improvement/changes in the documentation or code. I'm going to refer you to a forum post already made there on this topic:

closed 2019-05-16 by ericjunkins