nasa-jpl/open-source-rover #116: Monolithic board Electrical Assembly Instructions errors

JHPHELAN, 2019-05-18:
Electrical Assembly Instructions OMITTED D1 and F1.
They're clearly installed in the photos, but no reference in the instructions.  
They're the last left-over parts and slots prior to placing the standoffs.

On the PCB J13 (GND +12V) is omitted from the instructions.
On the BOM, J13 is the 5.08mm 2P side entry terminal block.  
But that is too large for the holes which match the 3.5mm J6 for the Arduino shield
(not to be confused, but it is, with the J6-7 2x20 40P ribbon cable connectors.)
There are 3 slots on the PCB for the 5.08mm 2P connectors labeled J15 (BATTERY IN), J16 (POWER SWITCH) & again J16 (BATTERY OUT).
Not sure what this J13 is supposed to do, but one of the two 3.5cc 2P terminal blocks is going into it.

Jim Phelan
Mars rover project lead
USAi Labs (formerly Houston Robotics)

[image]
[image]
[image]
[image]
[image]

ericjunkins, 2019-06-10:
@JHPHELAN Sorry I Must have missed this issue. Solving this currently by adding the instructions for which piece goes here. It is just a breakout of the 12V regulator line. I'm sure you had figured that out by now.

ericjunkins, 2019-06-10:
Fixed in commit 4b494c4

closed 2019-06-10 by ericjunkins