nasa-jpl/open-source-rover #117: Optional CAD parts heat-set-inserts #4-40 not #6-32?

JHPHELAN, 2019-05-18:
The Optional CAD parts lists a number 6 heat-set-insert for the new Arduino and back plate head mounts.  I bought both 6-32 and 4-40 for future use.  With some trepidation I tried the new technology on an old scrap rover back plate.  The holes were just the size of the 4-40 insert and too tight for the 6-32.  I drilled open one hole a couple sizes to just fit the narrow end of the 6-32.  Applying a needle tip soldering iron they both eased into place.  The 4-40 insert slipped out when pulled by a screw as the hole was a bit too large.  The 6-32 held nicely.  On the Arduino head, the 3D printed holes appear to better fit a 4-40 insert than the 6-32.  I'll wait for clarification on this before applying the soldering iron!

Jim Phelan
Mars rover project lead
USAi Labs (formerly Houston Robotics)

[image]
[image]

ericjunkins, 2019-05-18:
The head-base CAD part calls for M2.5 inserts actually. I can eventually change these to be 6-32 or 4-40 inserts to be consistent with some of the other things bought. I generally am using metric stuff at work at so my stockpiles are generally better for metric. The arduino plate would also be made for M2.5 screws/inserts, though I believe you could get away with 4-40s in them, they should be pretty close. In general the insert will not just push into the material, it should be melting away a bit of the walls as you push it in such that it actually is inset into the plastic. Because of this M2.5 and 4-40 are pretty similar, and M3 and 6-32 are also pretty close when it comes to the inserts.

ericjunkins, 2019-07-26:
Fixed in 71b9e5356ee9e42fb068ee36b2e46d12b22ea870

closed 2019-07-26 by ericjunkins