nasa-jpl/open-source-rover #118: Monolithic board testing fails at step two!  Test pad T2 isn't GND

JHPHELAN, 2019-05-18:
**Electrical Assembly Instructions / 2 Component Integration and Testing / 
2.1 Power Distribution System:**

"1. Begin by powering the board. In order to do this we will be bypassing the switch and
volt meter, so this connection will look a little bit different than when you fully install
the board into the rover. Insert the red wire on the Tamiya Battery Connector E35
into the IN terminal on connector J16, and the black wire to the GND terminal on
the connector J15."

Do we _really_ want to connect the battery to two different terminal blocks, or is one of these a typo?  [apparently it DOES work once you realize that T2 is NOT ground but use the battery ground as reference]

[image]

"2. Using a Digital Multimeter (DMM) probe the voltage accross the test points T1 and
T2. These will tell you the voltage level that the board is at, which is the direct voltage
of the battery. Verify that from T1 to T2 reads a positive number, and is between 12V
and 16.7V depending on the charge state of your battery."

Doing this, there is NO voltage between T1 and T2 when voltage is present between the two wires.

Connecting the battery to the more logical terminals on J15 has a similar problem.  Even shorting the POWR SWITCH J16 terminals failed to produce a voltage.

[image]
[image]

Further continuity testing revealed that test pad **T2 DOES NOT CONNECT TO GROUND**.  All the other GND terminals on the board DO test as connecting to the GND terminal of BATTERY IN.  I got an error message trying to view the PCB in KiCAD so I can't trace the problem.  I'll have to skip this part and proceed under the assumption that T2 is mislabeled or not connected.  So far no mention in the directions about installing the 12V or 5V regulators so they are not installed at this point.

"INSERT IMAGE OF ROBOCLAW WITH WIRES IN IT AND ROBOCLAW CON-
NECTION TO BOARD"  Permission granted.

[image]

Jim Phelan
Mars rover project lead
USAi Labs (formerly Houston Robotics)

ericjunkins, 2019-05-18:
I guess when I say "GND" yes this is being loose with the term "Ground". It refers to Battery Ground, or the (-) terminal of the battery. This is not Earth Ground.

Below shows the schematic of the board.
[image]

I will have to dig in further to that test point, i'll try and verify with my boards.  What is the error message from KiCAD? I want to make sure people can view/edit these boards so I want to try and help fix that as well.

JHPHELAN, 2019-05-19:
@ericjunkins 
I assumed that GND meant board GND / battery GND.  But T2 doesn't have continuity to  battery GND unlike every other GND designated terminal on the board.  However, if you ignore the problem with T2 and measure against any actual GND on the board, it measures ok so far as I've gotten.

The obviously old (as it bears little resemblance to yours) schematic I have (Monolithic Board.pdf) shows:
[image]

But I was less concerned about the labels as the actual board traces.  I may well be misusing KiCAD as I am not at all familiar with it, but when I tried to look at the actual construction of the PCB (Monolithic Board.pro) in KiCAD I got:
[image]

Jim Phelan
Mars rover project lead
USAi Labs (formerly Houston Robotics)

JHPHELAN, 2019-05-19:
BTW, my teacher at TXRX makerspace for Python & OpenCV, Dan Mailman, taught us that documentation always takes a back seat to creation.  EXPECT that documentation will be incomplete, out of date and inaccurate.  Just be pleasantly surprised when it actually works as hoped, just like your program.  That attitude has saved me a LOT of frustration!
JHP

ericjunkins, 2019-06-10:
@JHPHELAN I'm trying to close some of these open issues one by one here. I should expect that T2 is always GND for the system no matter what. Were you still seeing issues where it was not? I wouldn't expect that to happen and am just trying to check and see if this still needs to be an open issue or not.

JHPHELAN, 2019-06-10:
On my board T2 did not connect to anything.  Continuity check between it and  any other GND tests OPEN.
JHP

ericjunkins, 2019-06-10:
Ah okay, I found the error.. the following test point is not actually connected into the net.
[image]

leading to this via/net not being connected into the GND plane.
[image]

Fixed here:

[image]

commit 615c1514 (2019-06-10): Closes #118 by fixing gnd trace for T2 test pad
closed 2019-06-10 by ericjunkins