nasa-jpl/open-source-rover #119: M2A & M2B on the PCB are reversed vs the RoboClaw

JHPHELAN, 2019-05-18:
On the monolithic controller PCB, the labels for the terminal strips that jumper to the RoboClaw at M2A and M2B are **reversed**.  M2A is to the outside of the RoboClaw whereas the terminal strip outside connector is labeled M2B.  I hope it's just the labels that are backwards and not the internal wiring.  Else the wires between each RoboClaw and the PCB for M2 with have to be switched at one end.

Jim Phelan
Mars rover project lead
USAi Labs (formerly Houston Robotics)

[image]

ericjunkins, 2019-05-18:
The motor terminals do not particularly make a difference which is which for A and B, you can reverse the polarity during calibration such that it applies voltage to make it correct for "forward" of the rover

JHPHELAN, 2019-05-18:
All 5 RoboClaws power up!
JHP

[image]

ericjunkins, 2019-05-18:
Looks great! I do appreciate the feedback and pictures with your process putting this together. I know we released it at a very prototype-level stage and appreciate those of you who are helping test documents etc.

JHPHELAN, 2019-05-19:
2.4:
"INSERT IMAGE OF ROBOCLAW WITH WIRES IN IT AND ROBOCLAW 
CONNECTION TO BOARD"

Permission granted.
Jim Phelan
Mars rover project lead
USAi Labs (formerly Houston Robotics)

[image]

vssystemluba, 2019-06-13:
Thanks for pointing this out.  Even though it will be calibrated either way, I went ahead and flipped the silk screen labels (d2542d7) to keep others from getting confused.

closed 2019-06-13 by vssystemluba