nasa-jpl/open-source-rover #122: Please post photo of assembled Arduino shield

JHPHELAN, 2019-05-19:
@vssystemluba @ericjunkins 

Could y'all please post a pic of the assembled Arduino shield?  Most of it seems obvious but I need to clarify before applying solder.  I'll trade you a step-by-step album!  Thanks!

Jim Phelan
Mars rover project lead
USAi Labs (formerly Houston Robotics)

ericjunkins, 2019-05-20:
Will do when I get a moment thanks.

JHPHELAN, 2019-05-21:
I _believe_ it should look like THIS except I'm unsure about the orientation of the 3 connectors.
The power connector is too dark in this web view but is facing the far edge of the board
(you can save it and open it on your own computer to see it properly) :

[image]

Jim Phelan
Mars rover project lead
USAi Labs (formerly Houston Robotics)

ericjunkins, 2019-05-21:
As long as that 6 pin JST connector matches with the cable orientation you installed in your control board it is fine. See below pictures of where the 12V line is in both of them.

[image]

[image]

It also occured to me that the silkscreen is backwards on the J1 connector on the arduino shield. That was initially in a different location and upon moving it the silkscreen didn't get updated correctly. So that is mirror flipped as you can see below

[image]

JHPHELAN, 2019-05-22:
@ericjunkins Based on your circuits above, I assembled the Arduino shield as below.  
The only difference to my sample above is reversing the JST connector to face outward.  
Looks like we need a Rev 2 of the PCBs with the silkscreen reversals corrected & showing the key notches on the silkscreens to avoid confusion.  If there was a way to indicate on the PCB which way the header pins go, that would help too.

[image]

[image]

[image]

[image]

Jim Phelan
Mars rover project lead
USAi Labs (formerly Houston Robotics)

vssystemluba, 2019-06-13:
@JHPHELAN thanks for helping to add clarity here.  I've pushed a change to the silk screen (d2a88c6) that flips the labels which hopefully make things a little clearer.

closed 2019-06-13 by vssystemluba