nasa-jpl/open-source-rover #123: Monolithic Control Board right/left backwards??

JHPHELAN, 2019-05-19:
The revised Body Build Doc.pdf shows the monolithic control board hanging from the top of the rover with the R Pi hang down.  The BATTERY IN is near the Differential Pivot Block.  (An issue is noted with conflict with the neck's PVC mounting hub but I solved that just my moving the neck mount one hole closer to the center.)  In this position with the Differential Pivot Block at the rear and the neck at the front, the motor terminal blocks are on the OPPOSITE side of the rover from their labeled motors.  
Now what?  
a) ignore the R/L reversal as a silkscreen typo, everything is actually fine inside the PCB & software. 
b) ignore the R/L labels, wire the motors to the appropriate terminal block on the same side as the motors and handle the reversal in an upcoming software release or, please no, 
c) criss-cross the wires to the opposite side of the board.  
Waiting to tune the RoboClaws until this is clarified.

Also noted the U channels that used to support the old Electronics Board have been replaced by Dual Side Mount A's.

Jim Phelan
Mars rover project lead
USAi Labs (formerly Houston Robotics)

ericjunkins, 2019-05-20:
The silk screens are not a type on that case, the traces are in fact what they are labeled. But you should do option B). Polarity of the motor will make no difference and you will have to go on a motor-by-motor basis and verify that 'forward' command for pwm is 'forward' for the robot anyway, so just choose the correct polarity then

JHPHELAN, 2019-05-20:
@ericjunkins   It's not just a matter of polarity of the motors.  The entire right side of the rover is wired to the left side of the board and vice versa.   
JHP

ericjunkins, 2019-05-20:
Oh sorry I am mixing up what we were taking about... What do you mean it is left right flipped? That is a matter of what you're calling left /right from facing the rover or facing with the rover is it not? I did not criss cross the wires across the body, and my board is installed with the battery terminals near the differential pivot

JHPHELAN, 2019-05-20:
@ericjunkins 
I mean from the point of view of the rover itself as it is moving forward.  The terminals (J20, J21, J22, J25, J26) that are labeled for the motors (motors 1,2,8,9,&10) on the RIGHT side of the rover are located on the LEFT side of the PCB as it is mounted in the rover, and vice versa.
JHP

ericjunkins, 2019-05-21:
The J20,21,22,25,26 connectors should go to the right side of the rover, from the rover's frame of reference. To be fully clear I have boxed the motors I would expect that to control below:

[image]

This appears on the left of the board, because the board is put into the robot upside-down, such that the Motor controllers which are on the "bottom" of the board will face upwards in the body, and the Raspberry Pi which are on the "top" of the board face downwards in the body. Does this answer the issue here?

JHPHELAN, 2019-05-21:
No, that is exactly the issue.  In that position it seems the terminals for those motors are on the opposite side of the rover.  
JHP

ericjunkins, 2019-05-21:
Right, okay. I'm on board now. The silk Screen for left should say right and vice versa. 

[image]

Here is the PCB layout of the board. You can trace roboclaw to motor terminal here, but it seems that the left/right is upside down on the silk screen only, and that the traces go to the correct terminals

JHPHELAN, 2019-05-21:
Phew!!  Now I can proceed toward installation!
Thanks!!!
JHP

vssystemluba, 2019-06-13:
Thanks @JHPHELAN for pointing this out.  I've swapped the silk screen labels so they are consistent with the actual traces (d2542d7), and hopefully we saved other groups some confusion in the future!

closed 2019-06-13 by vssystemluba