nasa-jpl/open-source-rover #124: Electrical Assembly Instructions omit OpAmps until corner motors fail

JHPHELAN, 2019-05-19:
Electrical Assembly Instructions 2.3.3 
"...If that is not recheck that the OP-amp is installed in the correct direction."
There is no mention of the Op-Amps prior to this.  I assume they should be installed after the sockets for them?  Note that the key notch on the chip should match the key notch on the socket which, contrary to the directions, DOES matter which way it goes in if this convention is to work.

Jim Phelan
Mars rover project lead
USAi Labs (formerly Houston Robotics)

ericjunkins, 2019-05-20:
Roger that, it should have had that step before then. That picture is the correct orientation. I actually did not realize the dip sockets had a notch, in practice you can put the dip socket in either direction but i'll make sure to note it should be this way.

ericjunkins, 2019-06-10:
Fixed in commit 7c0e027

closed 2019-06-10 by ericjunkins