nasa-jpl/open-source-rover #126: Change second image to RPI with ribbon cable and usb cable plugged in

JHPHELAN, 2019-05-23:


JHPHELAN, 2019-05-23:
Ordering a MUCH shorter 7" USB-micro USB cable for R Pi power.
BTW the lighted push-button power switch is:
[image]

[https://www.digikey.com/products/en?keywords=GPB553B2L01BR](url)
JHP

billydelectronics, 2019-06-08:
I found a USB A to USB Micro 90 6" cable on Amazon that works SLICK for the Monolithic PCB to RPI power: https://www.amazon.com/gp/product/B00S8GU03A/ref=crt_ewc_img_dp_2?ie=UTF8&psc=1&smid=A1AMUYYA3CT6HJ  ASIN: B00S8GU03A
[image]

BillyD

ericjunkins, 2019-06-10:
Fixed in commit 7c0e027

closed 2019-06-10 by ericjunkins