nasa-jpl/open-source-rover #127: 3D printed battery holder doesn't fit battery

JHPHELAN, 2019-06-08:
Just printed the battery holder but find it doesn't fit.  Has anybody else tried this?  Made a fix?
JHP

[image]

ericjunkins, 2019-06-08:
oh, I forgot that I have a slightly higher capacity battery in mine so it is a bigger battery. Can you measure the dimensions of the battery and I can give you an updated part for it?

JHPHELAN, 2019-06-08:
@ericjunkins 
[https://www.batteryspace.com/li-ion-battery-14-8v-5-2ah-77wh-8a-rate-for-diving-light---un-38-3-passed.aspx](url)
"Dimensions  1.6"(43mm) H x 1.6"(43mm) W x   5.5 "L(139mm)"
I measure from the 2 flat sides, avoiding the bumpy wire, side **1 15/32"** a little less than the specs.
If the top of the holder were lowered to an inside jaws clearance of 1 15/32" it would be good.

JHP

ericjunkins, 2019-06-08:
yeah the dimensions of the datasheet aren't usually quite correct for these, they tend to be a bit bumpy and differently shaped. I'll upload a fix of that part

JHPHELAN, 2019-06-09:
@ericjunkins BTW the bottom holes are too big for 4-40 and just a little small for the 6-32 heat set inserts.
The 6-32 would probably melt its was in but the narrow end wouldn't have a clean start.
JHP

ericjunkins, 2019-06-09:
I've had no issues putting #6-32 heat set inserts in these parts

ericjunkins, 2019-06-21:
@JHPHELAN I finally got around to making edits of this for the 5.2mAH battery size. Can you let me know if this fits well or if it needs tuning still? It can be found at https://github.com/nasa-jpl/open-source-rover/blob/electronics-pcb-rework/Optional%20CAD%20parts/STL/battery%20holder_5_2maH.STL

ericjunkins, 2019-06-24:
My part for the test fit of this can be seen below:

[image]

I believe this fixes this issue and will close it based on commit b441a83c82498513b61260dd79068bb3b2afc41f

closed 2019-06-24 by ericjunkins