nasa-jpl/open-source-rover #128: Arduino IDE won't upload with shield in place

JHPHELAN, 2019-06-10:


ericjunkins, 2019-06-10:
@JHPHELAN Arduino IDE will not upload while you have the serial cable plugged into the Arduino sheild correct. This is because that uses the same serial peripheral on the Arduino for the upload as is used by the TTL serial communication line to the Raspberry pi. The arduino Uno only has one serial peripheral so there isn't really much that can be done about that.

I am thinking maybe adding a switch to the sheild to disconnect the serial Rx and Tx lines from the RPi would help though. So to upload code you would flip the switch, use the arduino IDE to flash, then flip the switch back to open communication to the RPi

JHPHELAN, 2019-06-10:
@ericjunkins ...even if the serial cable isn't connected to the Pi or anything elese?
JHP

ericjunkins, 2019-06-10:
The cable has electronics inside it that converts signals to the USB output whether it is connected to the Pi or not. Those are messing up the serial port on the arduino trying to flash

JHPHELAN, 2019-06-10:
@ericjunkins Thanks, but I'm still confused.
The RPi is not attached.  The 6-conductor jumper to the RPi is not attached to the shield.  Only the ribbon cable and power wires to the LED display are connected to the shield.  By "serial cable" do you mean "USB cable"?  It doesn't connect to the shield, but to the Arduino.

What I believe you mean is that the serial electronics in the shield interfere with the USB signal between the laptop and the Arduino if the shield is plugged into the Arduino.  This would explain the behavior.  But a switch disconnecting the Pi wouldn't help as I'm not connected to the Pi.  Perhaps a switch disconnecting serial communication between the shield and the Arduino might?  It is a pain to have to detach the shield every time I want to upload a revised sketch to the Arduino and risk damaging the shield.

Fortunately this won't matter for running the rover as the Arduino will just have its one sketch to pass the display info from the Pi.  For playing with just the Arduino and the LED display I could replace the shield with the "spaghetti farm" of jumpers between the Arduino and the LED display.
JHP

ericjunkins, 2019-06-10:
Okay wait sorry maybe I misunderstood. You should not ever have to remove the arduino shield. I believe these three desync error is caused by having both the USB and serial cable plugged into the RPi while trying to upload. I never had any upload issues if I simply unplugged the serial cable before beginning the flashing process

ericjunkins, 2019-06-10:
To be clear about the upload process (i'll soon be adding this to the instruction set) the following procedure is what I did to develop that library, which I probably ended up flashing that arduino on the order of a hundred times while writing that code.

*Assuming that the arduino sheild is installed on arduino, the 6 pin connector is plugged into the sheild, and the control board on the other side, as well as the ribbon cable to the LED Matrix*

1: Connect USB Cable from Laptop to Arduino Uno 
2: Unplug 6 pin JST connector from Arduino Sheild
3: Flash Arduino Code
4: Plug in 6 pin JST connector into Arduino Sheild

I would expect this to work on your board. Can you let me know if this is not?

JHPHELAN, 2019-06-10:
[image]
This is my setup.  NO RPi.  NO serial 6 conductor jumper. Just Laptop, USB cable, Arduino, shield and LED display connected by power and ribbon cables only.  If I remove the shield from the Arduino, the sketch loads ok &  runs the LED display when re-attached.  If I do not remove the shield and try to upload a sketch to the Arduino from the laptop I get the following error message (pardon the verbosity but better too much than not enough information).  Clearly something about the shield itself is causing communication problems between the laptop and the Arduino.  Disconnecting the ribbon cable between the shield and the LED display doesn't help. ---

JHP

Arduino: 1.8.9 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Me\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10809 -build-path C:\Users\Me\AppData\Local\Temp\arduino_build_926992 -warnings=default -build-cache C:\Users\Me\AppData\Local\Temp\arduino_cache_10073 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Me\Documents\Arduino\testshapes_16x32_PHELAN\testshapes_16x32_PHELAN.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Me\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10809 -build-path C:\Users\Me\AppData\Local\Temp\arduino_build_926992 -warnings=default -build-cache C:\Users\Me\AppData\Local\Temp\arduino_cache_10073 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Me\Documents\Arduino\testshapes_16x32_PHELAN\testshapes_16x32_PHELAN.ino
Using board 'uno' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\Me\\AppData\\Local\\Temp\\arduino_build_926992\\sketch\\testshapes_16x32_PHELAN.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Users\\Me\\Documents\\Arduino\\libraries\\RGB_matrix_Panel" "C:\\Users\\Me\\AppData\\Local\\Temp\\arduino_build_926992\\sketch\\testshapes_16x32_PHELAN.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Users\\Me\\Documents\\Arduino\\libraries\\RGB_matrix_Panel" "-IC:\\Users\\Me\\Documents\\Arduino\\libraries\\Adafruit_GFX_Library" "C:\\Users\\Me\\AppData\\Local\\Temp\\arduino_build_926992\\sketch\\testshapes_16x32_PHELAN.ino.cpp" -o nul
[... cut ...]
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x06
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x06
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x06
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x06

avrdude done.  Thank you.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

ericjunkins, 2019-06-11:
Okay I'm trying to figure out why this would happen. I'm assuming you have unplugged/replugged in your USB cable to the arduino after having removed the 6 pin JST connector?

Here are a few images of the pcb and the schematic of the board.

[image]

In the schematic we can see that the only thing that happens to the Tx/Rx lines for the serial read/write are that they run directly to the 6 pin JST connector at J5.

[image]

Highlighted is the board traces of these lines. This should entirely act as though there is just a wire attached to each of them that goes nowhere. 

Can you do a few things to help test this? Remove the Arduino sheild from the back of the Arduino and do the following:
 1. Connectivity test your serial pins. Resistance test from pin 4 on J5 connector to the top of the pin you soldered to for the Tx line (Arduino pin 1 https://www.circuito.io/blog/arduino-uno-pinout/). And then again from pin 5 on J5 connector to Rx line (Arduino pin 0). Make sure these values are a very small amount of ohms
 2. Connectivity test between Tx/Rx pins. Make sure there is no short between them. If there is a short this would surely cause the error you are seeing.
 3. Connectivity test between each of the Tx/Rx pins and all the power pins, the 12V, 5V, and GND. Make sure there are no shorts here as well. Shorts in any of the above would also cause a sync error.

I never once removed the Arduino sheild after inserting it onto my arduino, and can't come up with any reason other than a short on the board to cause this issue. Doing some googling about that error it is almost entirely caused by there being something going on with the Tx/Rx pins on the arduino. Let me know if you do these and still don't have anything conclusive

ericjunkins, 2019-06-11:
Also @JHPHELAN I edited your above comment to give it a collapsed set of text on all of the compiler and linker information of the Arduino. A nice way to have that there but not clutter up the message :)

JHPHELAN, 2019-06-13:
@ericjunkins SOLVED - hardware issue.
Thanks for the schematics and PCB outlines!
[what version of KiCAD are y'all using?  I suspect 4 and I have the latest 5 and there are compatibility issues which is why I have been unable to open them]

I tested TX and RX.  They were not shorted to each other and connected to the appropriate RPi and Arduino pins.

However, during the initial build of the shield, I had the serial connector backwards and had to rework it.
In the process I damaged the board as the GND line from the Pi did not connect to the GND for the LED display but was successfully jumped for a functioning display.  However, the 12V line from the Pi did not check as continuous with the TP1/12V nor the VIN pin of the Arduino.  Jumping that failed to solve the upload problem.

So, I rebuilt the board using one of the excess PCBs.  I didn't have a spare 16 pin connector so just used naked header pins and marked the side for the key slot.  Fortunately I DID have a spare 6 pin connector.
BINGO!!  I can now upload to the Ardino without disconnecting the shield!

JHP

closed 2019-06-13 by JHPHELAN

ericjunkins, 2019-06-13:
@JHPHELAN I'm using 5.0.2

[image]

If you are having KiCAD problems can you open an issue for that so we can track that? Want to make sure we don't miss something.