nasa-jpl/open-source-rover #131: Updated Software steps ttyS0

JHPHELAN, 2019-06-20:


vssystemluba, 2019-06-20:
Hm, that's interesting.  Are you using a Pi 3 for sure, or did you grab a Pi 2?  I believe the ttyAMA0 and ttyS0 swapped between Pi 2 and Pi 3.

The second thing to check is that your `sudo raspi-config` steps match those outlined in the Software Steps.pdf document.  Specifically, make sure your login shell is NOT available over serial (set to "No"), and that the serial port hardware is enabled (set to "Yes").

The short story is that you need to use `/dev/ttyS0` (so long as ttyS0 is mapped to GPIO pins 14 and 15 and not the bluetooth device).  The above `/dev/serial0` and `/dev/serial1` are aliases to the actual hardware.  If you've verified that the raspi-config configuration is correct and you've added the `console=tty1` to `/boot/cmdline.txt`, you can go ahead and try to do the serial comms.  If something fails, you may need to revisit this issue.

JHPHELAN, 2019-06-21:
It’s a Raspberry Pi 3 Model B+
The serial login shell is disabled
The serial interface is enabled
console=tty1 is in place
Does it matter if the keyboard & mouse are plugged into the rover? (No, it doesn't.  I tried)
NEXT STEP:
How do I pair the X-box controller to the Pi?
Then what are the X-box controls for the rover?
JHP

ericjunkins, 2019-06-27:
@JHPHELAN did you still need help on this? I wasn't sure I though i remember answering this somewhere else, but am unsure if I did or not

JHPHELAN, 2019-06-27:
Since y'all are using Raspbian instead of Ubuntu I started from scratch with the Raspbian / ROS kiinetic / OSR build.  In this system the ttyS0 comes out as Serial0.
JHP

closed 2019-06-27 by JHPHELAN